gnea/grbl #250: Inhibit Y axis

DavidA50, 2017-08-13:
I am not a programmer, so I would rather have a cut and paste answer then having to go in and Edit this and that. The question is: Is there a simple way to inhibit the Y axis movement whe the Z axis is down and the command is G01 point A to point B. I have built a Saw wit a 10 inch blade for cutting sheets of plywood so when the Z axis is down, if the Y axis were to move, or if in manual jog, it would self destruct

jahnj0584, 2017-08-13:
manually disable the Y driver? Or just use actual commands to control it
and dont touch the Y.

DavidA50, 2017-08-13:
I would rather use a command.

neilferreri, 2017-08-13:
@DavidA50 Are you worried you will accidentally send a command to move the Y axis? Are you jogging X to cut the ply?

chamnit, 2017-08-13:
@DavidA50: You have an interest question. What you ask is not as simple as a cut and paste answer because there are a lot of edge conditions that need to be handled correctly. For example, how should a Y command be interpreted when Y is locked? Should it return an error? Or be ignored? If ignored, should the parser track the last Y command position or not? What happens when you unlock the Y axis? These questions are just as critical because it can lead to the machine moving in ways that you may not have been expecting.

DavidA50, 2017-08-13:
I am concerned of an accidental Y axis move command.  Chamnit, Your right, I had hoped it was a simple answer. I have operated various commercial saws that operate on GCode, but the code is hidden and all the operator sees is the graphic of the cut path and settings such as depth past 0, kerf compensation. I may need to find a way to inhibit it with a mechanical switch

ASHIT-AXAR, 2017-08-14:
mechanical disabling would not work, because if you send a Y command (and Y is disabled) then there would be a step missing. and when you re-enable the Y axis you should do a "Homing".
(or you should work with relative movement/positions)

i think you should currently type the commands in "console" provided in the sender program.

langwadt, 2017-08-14:
pretty much all CNC machines are arguably one accidental away move from self destruction ;)

I you really want to avoid code a mechanical switch that connects Y step pulses to a limit input (mind the polarity) could work

DavidA50, 2017-08-17:
Thanks for all the input. I have found out that 2 of the commercial saw manufacturers us a GCode command of M98 to call a macro that disables the Y axis based on what the operator sets as table zero and work height. My solution was to add 2 DPDT relays that connect Y to a 2nd step motor with the shaft not connected. when z is moved down it trips a switch that connects the spare step motor, disconnects the Y motor and connects the Y motor to 5 volts at 1 amp (For holding torque). seems to work well only time will tell. Maybe in time some type of similar command can be added to GRBL.

electrokean, 2017-08-17:
@DavidA50 it sounds like you are switching your motor coils from the driver.  If so, don't do that!  You will very likely blow up your driver.  Instead, just switch the step input to the driver betwen the grbl output and 5v or GND (depending on your driver type).  That will keep the motor energised but unable to move.

DavidA50, 2017-08-17:
Yes, I am switching the coils. I should have added that there is a timer included so both sets of coils are connected for 1/2 second before the relays switch. I am using a CNC Xpro V3 board.

chamnit, 2017-08-17:
@DavidA50 : Usually M98 is a subroutine call only. It's not explicitly a disable or lock stepper axis. I'm not exactly sure what the other saw manufacturers are doing inside of the M98 subroutine that makes it disable it. Perhaps they just go through a set of procedures to alter an IO line to disable the stepper driver, much like you are attempting. The procedure would include delays and an order of operations.

If I had to do it without altering the g-code, I would likely just use one of the M7 or M8 coolant control commands to toggle an IO pin, which would control the stepper enable pin on the Y-axis stepper driver. When enabled, it would keep the driver disabled no matter what Grbl was doing. Perhaps, inverting the coolant pin and an AND gate would work. For extra points, it would also tie in the axis step pulse IO pin into hard limits, so that Grbl will throw an alarm if it detects any axis pulse.

langwadt, 2017-08-17:
Any software controlled disable can be "forgotten", A switch in Z is the safer option, it prevents any Y moves when Z is down. But there is no need to switch in a different motor, just connect y-step to ground when z is down, afaict y-step is exposed on a header. 

tying y-step to hard limit might not work  if the step pulse is too short compared to the filter on the limit input

DavidA50, 2017-08-17:
I have the CNC Xpro V3 board. It has pads for step/dir on the board but I havn't been able to find any data that shows it would disable the on board chips. That would be a better route in my mind to disable the step pulse.

chamnit, 2017-08-20:
@DavidS50: Sure that would work too provided that a logic chip to disable the step pulses won't degrade the signal. Setting the disable pin on the stepper driver would work just as well and not be as sensitive to electrical noise since it can be well buffered. 

Either way, it's a moot point due to your XPro board being fully integrated. 

If you can find a reputable gcode description that locks an axis (not a 3D printer/Marlin gcode), I'll entertain installing it. Preferably from a major manufacturer. 

@langwadt : Good point. I suppose it would still work if you didn't have filtering and using the hard limit feature only for pulse detection on the y-axis. Dual purpose would likely not work if filter was present without isolating the pulse signal from it.

DavidA50, 2017-08-21:
Thanks everyone for your input. I was overthinking it as I didn't want to have error's or alarms, or self destruction. I have modified the Z-axis switch and connected it to an isolation board that disconnects the Z-axis step input from the board and pulls it low. Works very well as I can adjust the distance from the table plane to when Z is active/disabled based on material dimensions.

chamnit, 2017-08-21:
@DavidA50 : We all might be a bit guilty of overthinking it. Lol. Thanks for the update.

closed 2017-08-21 by chamnit