gnea/grbl #252: Rounding error? (repost from old version)

Avalonnw, 2017-08-17:
Not sure what's happening but when using multiple tool changes in a program (I do manual changing with probing after using bCNC) it all works ok on first tool change and probe, but on next one I get an error that requested move on Z exceeds machine limits. For some reason, it always tries to move 0.01mm or 0.001mm (not sure now) beyond limit. I have to break program and zero Z every time before tool change.
I asked this on bCNC git, they pointed towards this community, saying that it's something to do with rounding in grbl. Can anyone help please?

chamnit, 2017-08-17:
You'll need to describe in detail your procedure that causes this issue. If it can be reproduced easily, it'll help a lot.

Avalonnw, 2017-08-17:
I'll try to do a video in the next couple of days (I disassembled my CNC to rewire limit swithes).
I use grbl-mega 1.1e controlling a 3040 Chinese CNC router. bCNC is the sender software. Machine zero is at far-right top corner.
I haven't touched any accuracy settings anywhere.
So, when I want to run a G-code, I bring machine to X and Y where I want to zero work coordinates and set G54 there and Z is zero on top.
Upon seeing T1 M6 router moves to a predetermined tool change location, and I change the tool. After toolchange, it slowly goes down to a touchplate and sets tool offsets. After that it goes to machine Z=0 (! see my suspicion further down) and moves to the beginning of the toolpath. So, this move to Z=0 sometimes fails:
- if I did not zero Z axis before toolchange
- if this is not the first toolchange (which is essentually the first point, as I can't zero mid-program)

So my suspicion is, that Z travel to 0 after probing is not in absolute, but relative coordinates and if I start with Z=0, then do probing (with Z resulting -42.564 for example) the controller tries to move +42.564 and all's fine and dandy, but if I had Z=-10.012 at the beginning of the cycle, then probing at -42.564, then returning to 0 results in error because for some reason somewhere in the loop the request of +42.564 results in Z travel to 0.001 which is beyond machine limit.

Sorry, I know it's confusing and I think I confused myself here, but that's pretty much what's happening. As I said, I'll try to make a video soon. If you wand me to test something specific, please let me know and I'll include it in the video as well.

chamnit, 2017-08-17:
When this alarm occurs, is the position error always within a step? or half of a step? If so, it's operating within spec. If this is the case, I know where the problem is coming from, but it's not a critical issue. I would suggest that you not work so close to the soft limits.

Avalonnw, 2017-08-17:
I'm not sure about steps. It seems to request that extra 0.001mm...
How do you avoid getting close to soft limits if it always tries to get to Z=0 after probing?

chamnit, 2017-08-20:
0.001mm translates to 1000 steps/mm. I'll make a development note to accept soft limit checks within a half step for the next major version  but I won't be applying this fix for v1.1. There really isn't any space left to devote to small issues like this, when you can fix it yourself by programming z-1 rather than z0. All space is reserved for critical bugs only.

closed 2017-08-20 by chamnit