gnea/grbl #254: CNC Plasma 'OK to Move'

nmeek, 2017-08-18:
My Plasma sends an 'OK to move' signal when the arc has transferred to the work, indicating that it is ok for the CNC to traverse. Is there any way to make use of this signal with GRBL?
I can add a dwell to wait for some arbitrary time, but would prefer to use arc transfer signal to start the piercing dwell.
Ideally, my piercing routine (handled in post-processor) would power on the torch at piercing standoff height, wait for arc transfer, dwell for prescribed piercing delay, move to cutting standoff and proceed with lead-in. 
Would it be possible to use G38.2 at very low velocity to wait for probe pin signal, then continue with job without resetting the z work offset?

Any help would be greatly appreciated.

chamnit, 2017-08-18:
If it's a signal, you could perform a feed hold and tie the signal into the resume input pin.

MeJasonT, 2017-08-18:
would it be possible for him to use the spare analogue pin (assuming its an arduino based grbl machine)

MeJasonT, 2017-08-18:
Analogue 4

chamnit, 2017-08-18:
@MeJasonT : Sure, if he wants to do some coding.

nmeek, 2017-08-18:
I'm using the Protoneer Raspberry Pi CNC Hat, v2.58.
It has no GPIO header, but looks like A4,6 &7 may be exposed via test pad on the board.

I do have some coding ability, but was hoping that the G38.2 would be more elegant and then use the normal GRBL build.

Can you use the probing commands during a job? If so, it could be used to wait for an input.
Also, G38.2 will error if the 'ok' signal is never received.

chamnit, 2017-08-18:
@nmeek : G38.2 would work, but you'll have to correct for the movement afterwards. What you're proposing isn't any different from feed holding and resuming. The only differences are you're tying into the resume pin, rather than the probe, and you don't have a error. Either way, if you want specific behavior, you'll need to alter the Grbl source code yourself to do what you want it to.

closed 2017-08-18 by chamnit

nmeek, 2017-08-19:
Thanks for the comments everyone.

Although the feed hold/resume also works, I like that the G38.2 throws an error if the arc transfer does not happen (due to plasma power/air flow/standoff distance). The feed hold would wait forever.

Another up-side to using the probe function is that it keeps the standoff distance as high as possible during piercing, minimizing splatter on the tip.

I got it working with G38.2 by adding the following piecing operation in my post processor:
  
   (RUNNING POINT-PIERCE COMMAND)
   G0 Z0.25
   G38.2 Z-.1 F0.025
   G0 Z0.1

This is a hack, for now, as it sets the cutting standoff height at 0.1" regardless of what I had it set at in Fusion. 

If anybody is interested, I have attached the post processor that i am using for Fusion. I used M8/M9 (coolant) for Plasma power ON/OF & the Probe pin for the 'OK to Move' signal from the plasma cutter.

[GRBL- Plasma NAM.cps.txt](https://github.com/gnea/grbl/files/1236428/GRBL-.Plasma.NAM.cps.txt)

Toutoune86, 2024-07-26:
Hello ! I am very interested in this thread, although it is pretty old now. Did it work well ? Was there any change  on the postprocessor, that looked relevant ? Any issue or recommendation if I try a similar postprocessing edition  (with Sheetcam, not Fusion)