gnea/grbl #255: Constant Error 22 since updating

cawoodm, 2017-08-19:
Just updated to Grbl 1.1 and keep getting error 22 with G01.

Once I manually do a G01 with F it works from then on... and then the same issue starts with G02.

So what is up? Does GRBL 1.1 really have no default feed rate? How would I set it once without doing G01?

jahnj0584, 2017-08-19:
What sender are you using

gerritv, 2017-08-19:
1.1 no longer has a default Feed Rate. Set one in your gcode file, or better yet use a post that does this for you.
A line at top of file with just F<speed> is enough

PicEngraver, 2017-08-19:
You can also just set a feed rate in a startup block $N=.
ex: $N1=Fxxxx, when xxx = your desired feed rate.
This way works when you want to jog with no file loaded upon startup.

cawoodm, 2017-08-19:
Grbl Controller 3.6.1
It doesn't allow any preamble so it's a bit of a pain but I've had more success with it than UGS or Chrome's gcode-sender.

cawoodm, 2017-08-19:
$N1=180 returns error:1

PicEngraver, 2017-08-19:
You forgot to include the "F"  -  F180

closed 2017-08-20 by chamnit