gnea/grbl #256: CNC Shield GRBL Limit Switch not Working

cawoodm, 2017-08-19:
I'm using the ubiquitous "CNC shield" with GRBL 1.1 and an Arduino UNO clone. The shield has a section called "END STOPS" with 6 pairs of pins X+, X-, Y+,Y-,Z+,Z-.

My assumption was when (assuming $21=1) I short or unshort X+ the machine will stop moving in the positive X direction. That's not the case. It occasionally works but in general $21 makes grbl unresponsive. I've checked the pins of the switch - they are ground and high (5v) so I assume a pullup is provided.

What's the correct way to wire the limit/stop switches to a *CNC shield*?:
* Normal open/normal closed?
* Capacitors
* Pull up resistors?

PS: I read here that the X-limit has something to do with pin 9 on the Arduino but I can't access that because of the shield. It also doesn't seem to go to the END STOPS section above.

wterreb, 2017-08-20:
Hi,  I'm also using one of the "clone" CNC shields and it works correctly
for me.  However there was one trick....   I had to comment out the #define
VARIABLE_SPINDLE line in config.h so that the Z-axis end stop would be
on Digital
Pin 11.

I also added 1uF Ceramic capacitors to ground on each of the three inputs
to filter out noise. Not sure if it makes any difference or not. Just
always been using it like that and not getting any problems using it like
that.

Hope my advise was of any help.

Werner

chamnit, 2017-08-20:
There is a wiki page on wiring limit switches. Suggest starting there. Also Disabling VARIABLE SPINDLE will work but v1.1's laser mode requires it. The problem is due to the pin 12 and 11 switch to access the hardware PWM. Z limit is now on D12. You can carefully alter the CNC shield if needed.

closed 2017-08-20 by chamnit