gnea/grbl #259: Max pulses per second??

brakthehun, 2017-08-25:
I built a largish CO2 laser (800mm x 1350mm) and I bought Nema23 425oz steppers to run each axis. I picked up two high end stepper drivers that go down to 1/128 steps. I'm using 20 tooth GT2 pulleys (40mm/rotation) and at 1/128 step,, I'm doing 640 pulses per mm. 

My problem is,, when I try to go faster than 3500mm/m it returns garbled nonsense and continues past the movement. When I go 5000mm/m it locks the Arduino.

Is there a max pulses per second??
Would using a mega help??

jahnj0584, 2017-08-26:
Your driver's AND arduino have a Max pulse rate. I believe it's 40kHz,as I
just posted as an error for UGS. Reduce your micro stepping to 1/64 or use
a better suited belt to get the step size required. At 200 steps/mm you can
go up to 12000mm/min before locking up

brakthehun, 2017-08-26:
Do you know what the max of the Arduino is?? I'd like to get as much fine detail as possible.

chamnit, 2017-08-26:
It's stated in the front page of the wiki and on the main page. It's 30kHz step pulses. In certain circumstances, it can go faster, but it can start to become unstable.

closed 2017-08-26 by chamnit