gnea/grbl #260: Internal calls to feed/rapid override

dastultz, 2017-08-28:
Hello, I am customizing GRBL for my hardware, a Sanguino/644p CNC machine. I have a number of buttons and a slider (pot) for jogging and override. I have tapped into `protocol_main_loop` to run my code. I have jogging working great. I'm trying to issue override commands at the push of a button. I thought it would be pretty simple. I see in the ISR for `serial.c` that the incoming RT commands result in calls to `system_set_exec_state_flag`. So I tried that but it doesn't seem to be working. I thought perhaps things might get reset at the top of the next loop, so I tried calling `protocol_execute_realtime` after with no change in behavior. Consider this:

```
printString("feed coarse plus\r\n");
system_set_exec_motion_override_flag(EXEC_FEED_OVR_COARSE_PLUS);
protocol_execute_realtime();
```
The print statement is visible in the console, but the override does not take place.

What is the best way to trigger rapid/feed override from within GRBL (1.1f)?

Thanks.

electrokean, 2017-08-28:
What you're doing looks like it should work to me, but I've not tried this myself.  Sonny is probably one of the few with enough internal knowledge of the code to help, but he is very busy and can't devote much time to help with this kind of distraction.

As you're working on what sounds like a heavily customised version, which could be introducing various complexities we don't know about, I suggest you take the standard 1.1f code (preferably on a 328P) and make minimal changes to test just this one bit of logic.

electrokean, 2017-08-28:
BTW, if this is being triggered by button presses, have you implemented some kind of debouncing and anti-repeat handling?  Otherwise you could be setting that override flag many times before the button is released.

chamnit, 2017-08-28:
Thanks @electrokean. I am very busy, but I'm still trying to answer questions the best I can. 

@dastultz : What you are doing should work. It's not clear to me why its not. The only reasons it shouldn't is if the newly calculated override is no different from the current one, or you are calling the protocol_execute_realtime() function inside an interrupt (doesn't work this way).

dastultz, 2017-08-28:
I'm polling the buttons in the main loop, no interrupts involved, and I am debouncing them.

If you say it should work, then I'll keep at it. Give me a week to figure it out and I'll update the issue with my findings.

Thanks.

dastultz, 2017-08-29:
OK, I've found that the code does in fact work, but it's very spotty, so I must have missed it earlier. As mentioned above, I call my own service routine from `protocol_main_loop` where I poll buttons, a pot, and set lights. It works great for jogging, quite responsive. But once I start a cycle (Universal G-Code Sender), it seems my service routine is called FAR less frequently, presumable because the system is awfully busy running things but maybe because it's running mostly on timers and interrupts?

Certainly I could set up my buttons on interrupts, but debouncing is much more complicated (for me). Is there a safe timer interrupt routine I can tap into to call my service routine from?

chamnit, 2017-08-29:
Thanks for the update. Execution may not return to the main loop for a while. I can get held up by the planner, if it's full and has to wait for it to empty before adding the next line motion. See mc_line().

closed 2017-08-29 by chamnit