gnea/grbl #261: Alarm 3

tejaskevadia, 2017-08-28:
When i testing some big gcode file.it give this ALARM:3 after 2 to 3 laks line ,some time after 50 thousand lines. Any hardware of soft RESET is not applied. what another possible reason for ALARM:3?

electrokean, 2017-08-28:
What are your settings?
Alarm 3 means grbl was reset during motion
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#alarm-message
This is typically an indicator of you running too high a step rate - i.e. over 30kHz
It could also be an electrical problem causing a reset

tejaskevadia, 2017-08-31:
below is my setting
$0=17
$1=25
$2=0
$3=7
$4=0
$5=0
$6=0
$10=3
$11=0.010
$12=0.002
$13=0
$20=0
$21=1
$22=1
$23=3
$24=500.000
$25=5000.000
$26=250
$27=5.000
$30=1000
$31=0
$32=0
$100=20.038
$101=25.183
$102=200.000
$110=15000.000
$111=15000.000
$112=600.000
$120=400.000
$121=400.000
$122=400.000
$130=2000.000
$131=2000.000
$132=2000.000
I was testing on my own android APP .below is screen shot of my app.
>APP was stop at line no 503653 due to ALARM:3
>Same gcode & same setting it trigger ALARM:3 at random line no. some time at 2lakh & some time at 50 thousand only.
[image]

*Below is screen shot of Error LOG. in my APP.
[image]
>My app using  simple send-response protocol. error555 in above screen shot indicate my app has sent one more line due to grbl is in IDEAL state.(I noticed in send response protocol, many time status report is in IDEAL state).**WHY GRBL HAS SEND IDEAL STATUS BEFORE "OK" RESPONSE.?**

electrokean, 2017-08-31:
It really looks like electrical noise is triggering the soft reset pin. 
Try adding a 1k pullup resistor from Arduino pin A0 to 5V, and see if that helps.

zeevy, 2017-09-04:
if you use simple send-response protocol and distance to travel is very short, then most of the time grbl will in in idle state only.

zeevy, 2017-09-04:
Too many error555, i hope you are not sending both \r and \n at the end of each gcode command

chamnit, 2017-09-05:
For the record, there is no error 555. Error codes only go up to 38. Something is strange with your streamer. Also, alarm 3 only occurs if a soft reset is commanded during a job. Either your sender did it (try a reputable GUI) or you have electrical noise triggering it on the soft reset pin.

tejaskevadia, 2017-09-06:
Sorry for late replay as i was testing grbl stability on some big gcode file.
-->error 555 is for testing. it is generated by my GUI when grbl is in ideal state & still it doesn't replay OK for last line.(How it is possible?)
-->As i do not require reset pin .& to avoid electrical noise i made below  change in cpu_map.h (bypass reset pin)
 #define CONTROL_MASK      ((0<<CONTROL_RESET_BIT)|(1<<CONTROL_FEED_HOLD_BIT)|(1<<CONTROL_CYCLE_START_BIT)|(1<<CONTROL_SAFETY_DOOR_BIT))
Still getting same Alarm:3 after running 8 hour.

chamnit, 2017-09-06:
Do you have an Arduino clone? Does it have a Ch380 USB serial chip?

chamnit, 2017-09-06:
@tejaskevadia : Also, to rule out your custom streamer and any electrical influences, have you tried streaming with another GUI or the Grbl Python streaming script? Grbl often runs jobs of several million lines of code and much longer than 8 hours. It's known to be quite stable, so it's a high probability that it's something on your end. Not sure what yet.

closed 2017-10-03 by chamnit