gnea/grbl #264: M3 Command stalls Feedrate V1.1f 08-01-2017

dutchpatriot, 2017-09-01:
Hi,

I noticed that somehow some way in GRBL version V1.1f-08012017 the M3 acts weird. If I run the Jaguar.txt gcode file the feedrate, which is set to F1500 goes down to 0, stalls. But I concluded that with the same feedrate or whatever feedrate I set on the dragon1_0001.txt gcode file which is also a M3 Sxxx setting doesn't stall. After checking the gcode I can only conclude that it has to do with the G1 and G0 movements because the dragon1_001.txt gcode file has mainly G2 and G3 arc movements.

The Jaguar_M4.txt gcode file runs ok and the feedrate acts as it suppose to.

[dragon1_0001.txt](https://github.com/gnea/grbl/files/1270330/dragon1_0001.txt)
[Jaguar.txt](https://github.com/gnea/grbl/files/1270336/Jaguar.txt)
[Jaguar_M4.txt](https://github.com/gnea/grbl/files/1270343/Jaguar_M4.txt)

chamnit, 2017-09-01:
Please post your $$ settings and $I build info strings. And are you running a laser job?

dutchpatriot, 2017-09-01:
My $$ and $I are:

Grbl 1.1f ['$' for help]
[MSG:'$H'|'$X' to unlock]
$0=10
$1=5
$2=0
$3=0
$4=0
$5=0
$6=0
$10=1
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=1000.000
$25=1000.000
$26=10
$27=1.000
$30=255
$31=40
$32=1
$100=99.684
$101=99.900
$102=250.000
$110=1500.000
$111=1500.000
$112=87.000
$120=100.000
$121=100.000
$122=10.000
$130=190.000
$131=280.000
$132=200.000
ok
[VER:1.1f.20170801:]
[OPT:V,15,128]
ok

I've checked it, converted the jaguar logo in another piece of software, Inkscape 0.92 with laserengraver extension and that extension spits out G2 and G3 which have no problems with this version of GRBL.

Yes I am running a laser job.

With the G1 M3 Sxxx commands it screws up the feedrate and reduces to 0
With the G1 M4 Sxxx commands it works fine...
thnx

chamnit, 2017-09-01:
As the Grbl laser wiki page states, M3 operates differently than M4. It's unclear what you think is a problem. 

With a G1 M3 Sxxx command, you have no motion attached to the command. Grbl has to finish all queued motions up to this point to set the laser power at this exact point in the program. This behavior is clearly stated in the laser [wiki page](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Laser-Mode), where it talks about laser operation through consecutive motions.

You also have your step idle delay setting at a low value. I will cause the steppers to momentarily disable every time it stops. Set this to 255.

dutchpatriot, 2017-09-01:
> 
As the Grbl laser wiki page states, M3 operates differently than M4. It's unclear what you think is a problem.
> 

Than why do all former versions what they need to do and only this version act different?

The wiki is written before the last 08-01-2017 version. I know that M3 acts differently than M4, M4 acts dynamically so the laserpower is adjusted at cross points and bends. M3 is on/off resp. with the power on the S (Spindle) command.

`
G90
G21
G0 X36.62 Y18.09

M3 S200
G01 F3000
G01 X36.6100 Y18.0900 F3000
G01 X36.6000 Y18.0900 
G01 X36.5800 Y18.0900 
G01 X36.5700 Y18.0900 
G01 X36.5500 Y18.0900 
G01 X36.5400 Y18.0900 
G01 X36.5200 Y18.0900
G01 X36.5100 Y18.0900 `

Isn't doing what it does in former versions of GRBL. In the last version it reduces the Feedrate to 0, at least on the feedback it says 0, because the feedrate is still going, it's only incinerating the workpiece and would take hours and hours to finish a small job.

And bye the way, I only drop an issue request when I tried everything else. So I know now that is is a problem with M3 G1 combination commands, M4 has no probs...

chamnit, 2017-09-01:
Maybe you are referring to the bug fix in February 2017. All versions since then have had this same update. The reason why M3 changed because it fixed an issue with M3 not syncing when it was supposed to in certain conditions. By enforcing M3 to stop when not used with a motion, it ensured that the laser was always set correctly and exactly where it is told to in the program.  Otherwise an M3 command could turn on or off when the user wasn't expecting to. This is an especially dangerous situation when using a laser.

Please be aware that I don't change things just because I can. Especially with master versions. Only if there is a critical issue that can risk the safety of a user or a minor bug fix that improves reliability without changing Grbl's operation.

Again, I may be misunderstanding what you think is a problem. If you are implying that there is a new bug, I would need to see a video that illustrates the issue.

dutchpatriot, 2017-09-01:
> Maybe you are referring to the bug fix in February 2017

> 

I am not referring to anything actually. I noticed something, tried to solve it with what I could and came to the conclusion that only the last version acts like I describe. All former v1.1f versions do what they suppose to do with the M3 and M4 command. Only the 08-01-2017 version of V1.1f is doing this. 

M3 doesn't stop when not in a motion, it just reduces the motion to a dangerously low amount (as good as 0, but still moving) so that you burn the work piece.

Flash a GRBL v1.1f_08-01-2017, and load up the file I attached in the first initial post.  Jaguar.txt. This is the M3 feedrate reducer, while the second jaguar file, Jaguar_M4.txt, is exactly the same file but with M4 command and that does work fine...

chamnit, 2017-09-01:
When you say all former versions, do you mean former releases? There have been three since January. There have also been 9 commits to the master source code throughout the year. You'll need to be more clear on which ones are effected. 

If M3 is slowly moving when it shouldn't, then there is a problem. This wasn't very clear in your previous posts. For now, I'll reopen this issue until I get the time to test and reproduce your problem, which may not be for a while.

dutchpatriot, 2017-09-01:
M3 is moving slowly when it should move fast. On the gcode feedback it than states FEEDRATE 0, which implies a stop, but it does move. 1 mm a second or less, didn't measure that.

These releases works fine:
https://github.com/gnea/grbl/releases/download/v1.1f.20170131/grbl_v1.1f.20170131.hex
https://github.com/gnea/grbl/releases/download/v1.1e.20170114/grbl_v1.1e.20170114.hex

With these the M3 command in conjunction with resp G1 or G0 Xxx Yxx Fxxx do what is expected.

The last release the M3 command in conjunction with resp G1 or G0 Xxx Yxx Fxxx reduce feedrate to nearly 0. G2 and G3 no problem, those work fine.

dutchpatriot, 2017-09-01:
Also my stepper motors make an awful noise when moving M3 on G1 and G0 movements in the last release. Because the steps are so small and movement so slow, probably because my laser is belt driven instead of ball screw.

chamnit, 2017-09-01:
@dutchpatriot : I just noticed that you are trying to execute a laser job with super high resolution per g-code command. Around 0.01-0.02mm per g-code command. On top of that, you are trying to execute it at 1500mm/min to 3000mm/min. This is well beyond the capability of the Arduino 328p. You are trying to push 2500 to 5000 g-code commands per second into a limited size planner buffer with a controller that can process a g-code commands only up to 400-500 per second. The second problem is that the planner buffer will only have about 0.15mm-0.3mm of distance to work with any given time (15 blocks at 0.01-0.02mm per block). You can't expect it to accelerate at all within that distance. I'm very surprised that it worked at all. 

Your M4 job "works", but it's not what I would call well. It seems that the job can execute just fast enough to maintain a certain low speed within the planner. Nowhere near the programmed speed. The M3 job likely doesn't work because it has to do a little extra stuff to check running conditions to slow it down just enough to drag. 

The G2/3 arc job works because Grbl breaks up the arcs into line segments much MUCH larger. On the order of 0.1-1.0mm. Almost two orders of magnitude larger than your Jaguar jobs. I suggest that you increase the line size to 0.1-0.2mm per g-code block. You should see it run fine.

closed 2017-09-01 by chamnit

dutchpatriot, 2017-09-01:
@chamnit, I just figured it out and indeed thats it!

I figured it out because of manipulating another gcode job with G-code Ripper. This program manipulated all the G2 and G3 arcs to G1 commands. By accident I checked the box to do that, after looking at the code I noticed that there where indeed no feedrate stalls on that job.

Nice to know we both learned today, you that GRBL even takes jobs it can not handle and just by itself somehow figures out to handle it, and I learned that the technical part of the Arduino side is limited to what you just nicely explained.

And indeed the M4 job worked out but not according the settings, but than again, GRBL just takes it sucks it up and figures it out by itself!

Thank you very much for your time, effort and explanation.