gnea/grbl #265: Pull-off during homing

rodfrey, 2017-09-01:
Hi! I feel awkward posting this here, since it's certainly operator error. But it seems to be the best place?

I'm converting my laser (a GWeike 1200x900) to use grbl-mega.  The problem I'm having is that when my machine homes, it doesn't pull off the limit switches but instead issues ALARM:8.

I was having trouble with limit switch noise before, but I built the low-pass filter described on the Wiki and the noise problems are gone (verified with my oscilloscope.) 

Now, when I type $H the machine will move simultaneously in X and Y, find one switch, continue and find the second one, stop and issue the error.  The status line shows the X and Y pins triggered.

I've changed $25 and it does affect seek rate, so that's wired fine. The machine just doesn't want to pull off the limit switches.

Here's my config:
$0=10
$1=25
$2=0
$3=3
$4=0
$5=0
$6=0
$10=2
$11=0.010
$12=0.002
$13=0
$20=0
$21=0
$22=1
$23=0
$24=200.000
$25=750.000
$26=250
$27=5.000
$30=1000
$31=0
$32=1
$100=208.333
$101=208.333
$102=800.000
$110=5000.000
$111=5000.000
$112=200.000
$120=500.000
$121=500.000
$122=50.000
$130=1200.000
$131=900.000
$132=380.000

Thanks for any guidance!

chamnit, 2017-09-01:
Is your machine trying to move away from the limit switches, but doesn't clear them all of the way? If so, increase your $27 pull off distance setting.

rodfrey, 2017-09-01:
@chamnit Thanks.  No, it's not moving at all... that's why $27 above is at 5.0 rather than the default 2,

The axes stop moving when they hit the limit switches for sure... I've moved 1mm away from the limit switches (via G91 G0 X-1 etc) and when I home they move 1mm and stop. Still about 15mm away from the physical limit.

chamnit, 2017-09-01:
You say that you are using Grbl-Mega (next time post there instead of the mainstream 328p version). Can you provide your $I build info string? Unfortunately I don't have any idea why the homing cycle would fail to reverse. Your settings seem fine. Do you have anything unusual connected to your machine? High power drivers?

chamnit, 2017-09-01:
BTW, Grbl should move your Z-axis first. Did you change the default build? If so, list everything that you changed.

rodfrey, 2017-09-01:
@chamnit Sorry about posting to the wrong place!  I'll happily move this there if needed.

The drivers are chinese copies of geckos.  7amps I think.  Moving around the table with G0 and G1 works well though.  

I did change the default build to be just X and Y axes. This is the build string:

$I
[VER:1.1f.20170801:]
[OPT:VNM,35,255]

chamnit, 2017-09-01:
@rodfrey : Thanks. Are you using the RAMPS support option by chance?

rodfrey, 2017-09-01:
Not using RAMPS... not on purpose anyhow. That would be defined in config.h if it were defined?

chamnit, 2017-09-05:
Can you post your config.h file? Altering to file type to .txt will allow you to drag and drop here. Also post a few pics of your wiring into your arduino and limit switches.

rodfrey, 2017-09-06:
[config.txt](https://github.com/gnea/grbl/files/1282581/config.txt)

Thanks chamnit. Here's the config.h file. I haven't taken pictures of the limit switch wiring although I can do so.  GRBL is correctly reading the limit switches, and in fact the "approach" cycle in limits.c works fine - it hits one limit switch and locks that pin in axislock, then finds the other one and stops when they're both in axislock.  

It's just that it doesn't then perform any pulloff motion at all, and calls system_set_exec_alarm(EXEC_ALARM_HOMING_FAIL_PULLOFF) at line 262 of limits.c.

I haven't been able to figure out where the pulloff motion gets planned... the first bit at line 233 seems to plan the approach motion and then lock the axes when the limit switch changes, but I don't see where the pulloff happens.

rodfrey, 2017-09-09:
I don't know if this is a meaningful clue, but I added some calls to report_realtime_status() during the homing cycle.  I also set the pulloff to 100mm.

What I discovered:
- It hits the limit switches (which had machine position 0 as a test precondition)
- It initiated the pulloff... the machine paused for the same amount of time it would take to move 100mm in each axis.  Neither axis moved during this pause
- The pulloff error was fired, but machine position was at 100, 100.
- The realtime debug reported BF:255,35 throughout, until the point where it "thought" it had completed the pulloff (but hadn't moved) - then it was BF:255,34
- The error gets generated because it tries to initiate the seek but it's still on the limit switches.

Probably not helpful, but dropping that here in case it triggers a thought.

rodfrey, 2017-09-09:
Solved. My stepper enable line was floating. I was fooled because normal jogging using G0 worked fine.

Most embarrassed. Thanks for your time @chamnit .

chamnit, 2017-09-10:
@rodfrey : Not a problem! Thanks for sharing. Reporting mistakes are just as important as solutions, if not more.

closed 2017-09-10 by chamnit