gnea/grbl #268: Spindle Override has no effect

jahnj0584, 2017-09-06:
https://github.com/winder/Universal-G-Code-Sender/issues/730

chamnit, 2017-09-06:
Please post your $$ settings, $I build info, and conditions that I can follow to reproduce the issue.

jahnj0584, 2017-09-06:
Spindle starts with an M3 S--- command. Sending a different S value changes the RPMs.
Pressing the spindle over ride buttons in the GUI just send 0x9(a/b) to the console but the PWM voltage does not change (checked with multimeter).

Settings: http://imgur.com/a/psJHQ

chamnit, 2017-09-07:
@jahnj0584 : I meant to provide an exact step by step process. Like what is the S value you are using. What is the override and voltage at that point? What happens when the override changes. I ask because what you assume should happen may not be correct. For example, a spindle override > 100% at your max rpm will not change the output, since it's already at max.  Also consider that the Arduino output is sometimes not exactly 5V and your 4.7V may be as high as it goes.

jahnj0584, 2017-09-07:
I tried using negative over rides at 24k and +/- rides at 14k rpm. In both scenarios the PWM voltage stayed at 4.7v UNLESS I sent a new M3 S--- command. The over ride buttons in both directions and increments had no effect. 

I'm using stock GRBL config.h and haven't changed anything there yet. 

It's possible 4.7v is really 5,seeing as the 120v outlet from the wall measures 116 (or may be some resistance, who knows)

chamnit, 2017-09-07:
@jahnj0584 : Ok that is a lot more clear as to describing the problem. I checked the source code where that type of thing is handled. You may be right. It doesn't look like spindle overrides are being updated when Grbl is IDLE. It should work if it's running. I don't have time to work on this until next week, but I'll try to test and address this soon.

jahnj0584, 2017-09-07:
Wicked. I guess theres no real reason to try this without running a file, but I was just testing out my VFD wiring. At first I thought I had the PWM plugged in wrong. Good thing I tried the other M3 S command option first! I will test it tonight and make sure it works while a file is running too.

chamnit, 2018-06-14:
@jahnj0584:
@109JB:
Pushed a fix for the spindle overrides updating during an IDLE state. Should work. Should also operate correctly if laser mode is enabled (M3 only, M4 disabled until motion). Please try to break it and let me know how it goes.

closed 2018-06-14 by chamnit