gnea/grbl #271: Cannot flash with arduino IDE

jahnj0584, 2017-09-19:


neilferreri, 2017-09-19:
@jahnj0584 How did you modify the config?  Looks like you don't have a `#define HOMING_CYCLE_0`

chamnit, 2017-09-19:
This happens be if you don't clear out the files or if you add the wrong folder. Please follow the directions exactly.

closed 2017-09-19 by chamnit

jahnj0584, 2017-09-19:
i opened config.h in notepad++ and commented the first homing cycle, then changed the 1 to a 0 on the next line (for x y axis). Saved

Then added THAT grbl folder to the arduino IDE and tried flashing it.

neilferreri, 2017-09-19:
@jahnj0584 What are you uploading?  It looks like you have your own sketch.  Are you running the example, "GrblUpload"?  That is what you should be doing.  The grbl folder, with the modifications you made, should go in your "libraries" folder. Make sure there are no other instances of grbl in that folder.  The only thing you need open in the IDE is "grblUpload".

jahnj0584, 2017-09-19:
It's all fixed now. I was not doing step 4. It flashed properly and does home now with two axis. There were some errors in the arduino but all from files I never opened or touched.