gnea/grbl #272: Spindle PWM calibration

jahnj0584, 2017-09-19:
Any way to calibrate the PWM to match real world values? I only read 4.7v instead of 5 nominal, so the spindle is running 2-400rpm below what is is programmed to do. What is the best way to calibrate this so software matches the hardware?

chamnit, 2017-09-19:
Yes. That's what the max and min rpm settings are for, which assumes a linear relationship. If you need a nonlinear fit, there is a config option for that. See the Python script for how to set the constants for it.

closed 2017-09-19 by chamnit

massiveradiation, 2018-04-06:
I know this is closed. I'm new to python and I'm trying to setup the non-linear curve for my super-pid, but am having troubles. When I copy paste the python code into the web site given, I get errors. The first one is a syntax error on line "file main.py" line 26. Sorry, I've been looking around for someone to teach a newbie like me how to do this, but I'm coming up empty handed. I'd love to create a youtube video of the process if I can figure it out. Thank you for all the hard work you're doing on grbl! I use it all the time. Fantastic stuff. I just want to be able to enter an "S" value in g-code and have my pid go to the same rpm. I understand that the speeds to voltage output of the arduino are not linear of course and I get the concept of entering a series of input value vs real rpm based on tach readings. I just don't know how to get the script to run. If there is information other than the instructions provided, I'd happy to read through them if you can point me in the right direction. A summary of step by step would be so helpful. Example:
1. Ensure grbl v1.1f (latest version) installed and whatever settings need to be checked in config.h before upload.
2. Set $30=1, $31=255
3. Copy paste ??? to website and run python. If you get error, why... etc.
Again, thanks for humoring me. If there is another place to post this, Please let me know.
Cheers,
Tyler

chamnit, 2018-04-06:
I copy and pasted the fit_nonlinear_spindle.py  in the /doc/script folder into the repl.it webpage. It ran fine. If you edited the file, you may have introduced a syntax error.

massiveradiation, 2018-04-06:
Thank you for getting back to me so quickly! Hmmm. I didn't edit it. On that page, do I need to "add file," name it, then cut and paste into new file? Leave the main.py blank? Also, which version of the fit_nonlinear_spindle.py should I use, there are several. Can you give me a link? I think I have the latest one. It doesn't seem to like the two lines of "+" next after the """

massiveradiation, 2018-04-06:
Hmmm, maybe it's because it doesn't give me an option to choose, "Python REPL." Also, I'm pasting into a file, not sure what the "browse editor" is, as you've specified. Sorry, very green...

massiveradiation, 2018-04-06:
This one? https://github.com/gnea/grbl/blob/790c666ecbe0775019ee9deef51c27db33409ab9/doc/script/fit_nonlinear_spindle.py

chamnit, 2018-04-06:
Looks like they changed how repl.it works since the last time I used it. Copy and paste the raw text of the fit_nonlinear_spindle.py file into main.py. Create a dummy file in repl.it so that the generated image will appear after you run it. Just click the "Add File" button. Name the file anything. Doesn't matter. Just leave it blank. Then go back to main.py and click run. You should get some text output and an image file (png) will appear in the file list.

massiveradiation, 2018-04-06:
Aaaaaaah! Let me try...

massiveradiation, 2018-04-06:
YES! That was it! Now, I see the output. Thank you soooo much. One more question while I have your attention, there are other sections of the site, one referring to buffer underrun etc. Do I need to change my other grbl files with those? If so, where do I alter the file, place at beginning? Does that make sense?

massiveradiation, 2018-04-06:
And should my $30=1 and $31=255? Make changes only to cpu_map.h?

chamnit, 2018-04-06:
No, you're not making any sense. Default grbl should work fine. The only things you should change are standard $$ settings and in the config.h, which contains more advanced options like enabling the nonlinear spindle part of the code. You shouldn't mess with cpu_map.h or any other file.

Max and min rpm values are arbitrary. Set them to whatever you want. Usually you'd make them close to real world values.

massiveradiation, 2018-04-06:
Sorry if I'm not making sense. Here is my output with default. Down below it says to alter the cpu_map.h file:
Python 3.6.1 (default, Dec 2015, 13:05:11)
[GCC 4.8.2] on linux
   

CONFIG:
  N_pieces: 4
  PWM_min: 2.0
  PWM_max: 254.0
  PWM_point1: 20.0
  PWM_point2: 80.0
  PWM_point3: 150.0
  N_data: 15
  PWM_set:  [  2.  18.  36.  55.  73.  91. 109. 127. 146. 164. 182. 200. 218. 237.
 254.]
  RPM_measured:  [  213.  5420.  7145.  8282.  9165.  9765. 10100. 10500. 10700. 10900.
 11100. 11250. 11400. 11550. 11650.]

SOLUTION:

[Update these #define values and uncomment]
[ENABLE_PIECEWISE_LINEAR_SPINDLE in config.h.]
#define N_PIECES 4
#define RPM_MAX 11681.7
#define RPM_MIN 203.5
#define RPM_POINT12 6167.9
#define RPM_POINT23 9622.6
#define RPM_POINT34 10813.8
#define RPM_LINE_A1 3.017908e-03
#define RPM_LINE_B1 -1.385761e+00
#define RPM_LINE_A2 1.736766e-02
#define RPM_LINE_B2 8.712245e+01
#define RPM_LINE_A3 5.876519e-02
#define RPM_LINE_B3 4.854753e+02
#define RPM_LINE_A4 1.198347e-01
#define RPM_LINE_B4 1.145869e+03

[To operate over full model range, manually write these]
['$' settings or alter values in defaults.h. Grbl will]
[operate between min($30,RPM_MAX) and max($31,RPM_MIN)]
$30=11681.7 (rpm max)
$31=203.5 (rpm min)

[Update the following #define values in cpu_map.h]
#define SPINDLE_PWM_MIN_VALUE 2
#define SPINDLE_PWM_MAX_VALUE 254

PCPayne, 2018-04-06:
Hi Guys, i started a new issue few days ago about connecting my laser. Can anybody help me at all please.

massiveradiation, 2018-04-06:
Thanks again chamnit! I'll play around with it now based on what you've said.

massiveradiation, 2018-04-06:
Is there a reason there is a period after the 213 in the RPM array?

chamnit, 2018-04-06:
@massiveradiation : The period denotes a floating point value in Python. Otherwise, it can sometimes interpret it as an integer.

massiveradiation, 2018-04-06:
Shall I start getting router tach values with $30=254 $31=1? Then later, after config.h is changed with nonlinear adjustment, change to actual rpm values? In the sample array entries, it appears that's what's being done.

massiveradiation, 2018-04-06:
Got it working. Pretty damn close. I set $30=254 and $31=1 an collected 15 data points of rpm. For the record I tried to use about 20 data points and the results were whacky! Is this what is meant by using only 4 pieces? Not sure. After copy and pasting the results of the script, I changed $30 and $31 to min/max values listed in the results. It's good, but I think some more well thought out data points in the rpm range will yield closer results.
Thanks again for your help! I'll create a video of the process and post to my youtube channel, wiremonkey: https://www.youtube.com/channel/UCKwyc01hSo3kpFYGm77d0rw?view_as=subscriber

chamnit, 2018-04-06:
Adding more data points shouldn't change the fit algorithm. Can you share the setup parts of your python script and the plot? I'd like to see if there was something else that caused the problem. 

A piece-wise linear approximation with four line should be good enough for mildly nonlinear curves, especially if you set your intersection points well. Adding one or two more lines may be possible, but it isn't something that important for a general release.

massiveradiation, 2018-04-06:
> Sonny Jeon <mailto:notifications@github.com>
> April 6, 2018 at 4:34 PM

Interesting. Maybe I did something else wrong. It's easy to get lost 
when entering a ton of data. I'll try again. the result gave me a 
massive rpm min of over 600K!
Thanks again for this. So cool!