gnea/grbl #275: Using both NO/NC for limit switches

Mark-81, 2017-09-21:
Hi! I read through the documentation and other issue without found an answer. I apologize if I didn't search well! My CNC has already the limit switches connected with both NO (to GND) an NC (to 3.3V). The common is routed to the Arduino input. In this way the line is forced high or low and never "floats" (i.e. rely on internal pull-up).

Do you see any problem doing this?

jahnj0584, 2017-09-21:
Common goes to ground. NO or NC get wired to the relevant pin on the board.
You don't do both.

biasedlogic, 2017-09-21:
@Mark-81 Yes and no...
The switch wiring itself will work as expected - mostly.
The issue I'd have is with having long wires going directly to µC port, routed across an EMC-offensive environment such as a CNC mill (even a small one).

Mark-81, 2017-09-21:
Well, I understand it's not the best way, but I used that machine for years
with no problems.

Why keeping the input pin to a low impedance rail (gnd or 3.3V) is worst
than just rely on the internal pull-up?

biasedlogic, 2017-09-21:
> Why keeping the input pin to a low impedance rail (gnd or 3.3V) is worst than just rely on the internal pull-up?

I didn't say that using internal pull up was better...
If i were you, I'd put at least a resistor in series with the signal wire (between switch and pin), say 1k, and a 100nF cap between Arduino pin and ground. I would also put a 10k pull down direct at Arduino board, so that shall a wire break at the switch the whole thing will react

cncpaw, 2017-09-21:
(Qualified Electrical and Electronic Engineer)
Should/Can this issue me merged with "Limit switch wiring"?

NO+NC has no advantage in this case.
The limit switch to Arduino interface needs at least these things or equivalents:
1) A typical limit switch needs a wetting current of about 5mA (1kohm at 5V) to keep the contacts clean. The ATmega internal pull-up is nowhere near enough (about 60k), but it will function with new switches.
2) The ATmega needs protection against noise spikes, e.g. in the form of a series resistor (see 3), otherwise the spikes can damage the ATmega input.
3) Noise filtering, e.g. a Capacitor/Resistor low-pass filter, typically 10kohm & 100nF. Without this GRBL will trip spontaneously. A time constant (C x R) of about 1ms-10ms should do in most cases, long enough to remove the biggest spikes but not so long as to reduce the homing accuracy at the homing feed rate.

In addition these are highly recommended:

4) Screened cables so that noise on the switch wires is avoided in the first place.
5) Isolation (an opto-isolator) to get protection against spikes, ESD, ground loops and short circuits.
6) Fault tolerance, e.g using switches that are closed when the machine is out of limit. This is because the most common failure modes for switches and wires is open circuit. (Note: This usually means using the contacts marked "normally closed", except for a few CNC machines that keep the switched pressed while out of limit.)

In your case you can leave the NO contacts as they are and treat it like a NC to +ve. Use a 1k pull down (for 5mA @ 5V) and a 10k/100nF filter. All C's and R's on the arduino end of the cable. The capacitor could go to +ve so you don't get a transient limit, but I think GRBL is immune to that. Ideally the switch +ve supply should be the same voltage as the Arduino +ve supply for best noise immunity.
(NC gives lowest impedance in the non-limit state when GRBL needs it the most. But cable inductance can still let through high-frequency noise. I don't think low impedance in the limit state is as important)

closed 2017-10-03 by chamnit