gnea/grbl #276: Switching Laser On and Off.

StuartB4, 2017-09-23:
Cant seem to sort this.
With GRBL 0.9  I use a transistor to switch a relay which connects the laser power to switch the laser on and off.
It uses M3 and M5 in the GCode program.
I have the base of the transistor connected to pin 12 on my Arduino Uno via a 1k resistor.
When GRBL sends an M3 command it switches the transistor on which in turn switches the relay on and connects the laser power. when it receives an M5 command it switches the laser power off.

I cant do this in GRBL 1.1.
Any idea what I need to change or does 1.1 not allow this.

Thanks.

neilferreri, 2017-09-23:
The default for pin 12 is for the Z-axis limit. Pin 11 is the spindle enable pin.
https://github.com/grbl/grbl/wiki/Connecting-Grbl

StuartB4, 2017-09-23:
That does'nt really help. If I connect the transistor to pin 12 the laser comes on as soon as I connect to a com port.
If it's on pin 11 it does'nt come on at all.

jahnj0584, 2017-09-23:
Yes so use the spindle PWM as the control... M3 on M5 is off

StuartB4, 2017-09-23:
Fixed now. Changed line 339 in config.h to disabled.
// #define VARIABLE_SPINDLE // Default enabled. Comment to disable.

Thanks for the help.

closed 2017-09-30 by StuartB4