gnea/grbl #278: Homing with LaserMode (no Z AXIS)

GALGIGA, 2017-09-26:
HI.

i designed one piece printed XY plotter, and operates it with GRBL. having one issue - homing cycle. 
all limit switches are wired as it shown (accept Z axis - not wired at all).
everytime i run thr home cycle ($H) its return a failure message and restart the program.
my guess is, its have an issue with the Z axis. and i guess there is a simple solution, regarding its similar to LaserMode.

i maybe wrong, i lack of experience with the GRBL

thanks

X3msnake, 2017-09-26:
You have to disable Z flag

I Beleive it must be done on the config.h file before compiling to the
arduino

biasedlogic, 2017-09-26:
Yep, the config.h holds the homing sequence configuration. It's all well documented within the config.h file. The file itself is in Arduino/libraries/grbl directory.

GALGIGA, 2017-09-26:
thanks. i will check it
and how can i change it? 
mark the z axis line as comments?

biasedlogic, 2017-09-26:
@GALGIGA pretty much so. Comment out the 3-axis homing section and uncomment two axes. There are up to three steps of homing, I don't have the file in front of my eyes, but it's like that:
#define homing_0 (1<<AXIS_Z)
#define homing_1 ((1<<AXIS_X) | (1<<AXIS_Y)

which translates to: home only Z in first step, then home XY at the same time.

For my plotter I changed it to 
#define homing_0 (1<<AXIS_X)
#define homing_1 (1<<AXIS_Y)
so that the X axis is homed first and then the Y axis. You could do also:

#define homing_0 ((1<<AXIS_X)|(1<<AXIS_Y))
//#define homing_1 

to home XY at the same time (not good for my plotter for mechanical reasons).

Replace the exact wording with what you find in the config.h file, it's all human-readable

GALGIGA, 2017-09-26:
i changed the code. indeed it was pretty simple
now for some reason when i write the homing command its do nothing. and after few seconds i getting a fail msg (maybe due to lag time). ever happened for you before?

chamnit, 2017-09-26:
@GALGIGA : It's important to tell us what the error message is. It's likely the error code where it failed to trigger a limit switch during the homing search phase. Make sure you are editing the correct config.h located in the Arduino library folder, not the downloaded one. Also make sure your machine is connected and setup correctly.

closed 2017-09-26 by chamnit

biasedlogic, 2017-09-26:
1. Check your config, especially steps per mm, machine size and home seek speed. If these are wrong the step frequency may be too fast to high and the steps won't budge, or grbl will think it exceeded machine size while seeking.
2. Double check your wiring, add filtering caps and series resistors. I got all kinds of funny behaviour when I test-rigged my plotter to cnc shield without them, due to y-limit wiring running parallel to motor power in a drag chain.

jahnj0584, 2017-09-26:
Make sure your arduino library was cleared properly before reflashing

GALGIGA, 2017-10-02:
UPDATE: HOMING WORKS!! 
after few days i didnt dealt with the grbl. i just got to it and it works
thanks all of you for the help