gnea/grbl #279: gcode optimization rules for grbl

arkypita, 2017-09-29:
Hi @chamnit 
I am going to write a function to parse a gcode file and strip out all the redundant commands and parameters.

I know that some commands and params are modal: if you send S120 you dont need to send it again unless you want to change power, if you send G1 you can send a sequence of XYZ movement without sending G1 again etc

My questions are:
- This concept of modal commands and parameter are proper to grbl interpreter or they are defined as standards for any gcode interpreter?
- Can you confirm that grbl has two different set of memories for G0 and G1-2-3? As example i can set G0 F1000 and G1 F2000 on top of the code and all the next G0 will be executed with F1000 and G1 at F2000? Is this a grbl implementation or a gcode standard?
- Do you know any place where I can find some documentation about modal/non-modal command and parameters that can help me writing this function?

Thanks a lot, Diego

chamnit, 2017-09-29:
- Yes. They follow the LinuxCNC gcode standard but not all gcode interpreters are the same. Modal states are generally the same though. 

- G0 has no feed rate. It is a traverse or rapid motion. This is something that Smoothie and maybe Marlin incorrectly does. It’s due to a mistake in very early GRBL versions (before I started working on it).  No where in any gcode standard does it state that G0 can be programmed with a feed rate. This is primarily for safety because it would be easy for a user to confuse the two. 

- See the LinuxCNC documentation. They are very good.

closed 2017-09-29 by chamnit

109JB, 2017-09-29:
Most if not all of the modal commands are standard among interpreters/controllers, but there may be some differences.  The NIST RS274 NGC is the standard that Grbl started with. That standard is also what LinuxCNC uses at its core, but LinuxCNC has some differences from NIST to incorporate more functionality. Grbl has been following the LinuxCNC model for several years and the definition of commands that LinuxCNC has can be found in the linuxCNC documentation http://linuxcnc.org/docs/ The Grbl supported commands closely follow the LinuxCNC methods of implementation.

As for G0 F1000, in this line the F1000 will be ignored. G0 is a rapid traverse move and by its definition will move the machine at the maximum rate the machine is capable of regardless of the F word setting.  For G0 the F-word is meaningless. However, the F word would be read and then would change the feedrate for subsequent commands that use the F word (G1, G2, G3...)

There is only one F-word definition.  If you change the F-word value you change it for all commands that use it. This is standard across all g-code interpreters/controllers I am aware of.

http://linuxcnc.org/docs/ 

As for your project, I assume that this is an attempt to speed up Grbl's processing of lines of g-code.  @chamnit can probably be more specific, but I have also done some testing in this regard with my GUI and have come to the conclusion that it doesn't help much unless the g-code is littered with redundant commands and those commands are very small movements.  Most are not bad enough to see any type of significant gain.

winder, 2017-09-29:
Some particular sections which I found useful for getting my bearings when working on this problem:

[Format of a line](http://linuxcnc.org/docs/2.7/html/gcode/overview.html#_format_of_a_line)
[Modal group documentation](http://linuxcnc.org/docs/2.7/html/gcode/overview.html#_modal_groups)
[Order of execution](http://linuxcnc.org/docs/2.7/html/gcode/overview.html#_g_code_order_of_execution)

arkypita, 2017-10-01:
thanks to all of you