gnea/grbl #281: How to get the $SLP command to work.

StuartB4, 2017-09-30:
I use LaserGRBL with a genuine Arduino UNO with GRBL 1.1 installed and have Easydrivers to drive my motors.
There is a sleep connection on the Easydrivers which is set to normally high.
At the minute I don't have anything connected to it.

This is from Easydrivers website:
"SLP (sleep) : This normally high input signal will minimize power consumption by disabling internal circuitry and the output drivers when pulled low." 

What I'd like to know is do I need to link a wire from one the Arduino pins to the SLP connector to enable $SLP to work.

At the minute when I type the command in LaserGRBL it responds with:
[MSG:Sleeping] but does not actually do anything, everything is still powered including the motors.
[image]

Any help would really be appreciated.

chamnit, 2017-09-30:
Grbl's sleep command just disables the stepper enable pin, based on your $4 step enable invert setting. It's not tied to your stepper driver sleep pin.

closed 2017-09-30 by chamnit

StuartB4, 2017-09-30:
So do I need to connect the Enable pin on the Easydriver to the Arduino.
Because I've just made $4=1 and it still does'nt work.

On the Easydriver the Enable pin is this:
"ENABLE : This normally low input signal will disable all outputs when pulled high."