gnea/grbl #282: Feedrate reporting problem when both axis are moved

zeevy, 2017-09-30:
My machine max feed rate is 3600. 
When I move single axis G0X280 feed rate reported by grbl via ? query is 3600, however if I move both X and Y axis G0X280Y380 then feed rate reported is around 4472.
I have no idea why the feed rate is more than the max value 3600.

Machine settings

$0 = 10    (Step pulse time, microseconds)
$1 = 255    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 1    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 2    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 1    (Soft limits enable, boolean)
$21 = 1    (Hard limits enable, boolean)
$22 = 1    (Homing cycle enable, boolean)
$23 = 3    (Homing direction invert, mask)
$24 = 150.000    (Homing locate feed rate, mm/min)
$25 = 800.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 10000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 400.000    (X-axis travel resolution, step/mm)
$101 = 400.000    (Y-axis travel resolution, step/mm)
$102 = 200.000    (Z-axis travel resolution, step/mm)
$110 = 3600.000    (X-axis maximum rate, mm/min)
$111 = 3600.000    (Y-axis maximum rate, mm/min)
$112 = 2400.000    (Z-axis maximum rate, mm/min)
$120 = 240.000    (X-axis acceleration, mm/sec^2)
$121 = 240.000    (Y-axis acceleration, mm/sec^2)
$122 = 240.000    (Z-axis acceleration, mm/sec^2)
$130 = 285.000    (X-axis maximum travel, millimeters)
$131 = 385.000    (Y-axis maximum travel, millimeters)
$132 = 55.000    (Z-axis maximum travel, millimeters)

[image]
[image]

biasedlogic, 2017-09-30:
G0 does not observe a set feed rate, but only maximum axis speeds per axis. The reported speed is the 'tool holder speed', in theory about 5000 in your setup (x y both moving at 3600)

chamnit, 2017-09-30:
Grbl does a vector addition of the axes max speeds. For example, diagonal XY moves will be the sqrt(x_max^2 + y_max^2). Grbl will always maximize speed whenever possible.

closed 2017-09-30 by chamnit

zeevy, 2017-09-30:
Thanks for the clarification