gnea/grbl #284: Why did you close an issue before I've finished.

StuartB4, 2017-09-30:
I'm trying to get the $SLP command to work and asked for help, half way through the issue was closed.
That's not really very helpful. 
I started with this:

> I use LaserGRBL with a genuine Arduino UNO with GRBL 1.1 installed and have Easydrivers to drive my motors.
> There is a sleep connection on the Easydrivers which is set to normally high.
> At the minute I don't have anything connected to it.
> This is from Easydrivers website:
> "SLP (sleep) : This normally high input signal will minimize power consumption by disabling internal circuitry and the output drivers when pulled low."
> What I'd like to know is do I need to link a wire from one the Arduino pins to the SLP connector to enable $SLP to work.
> At the minute when I type the command in LaserGRBL it responds with:
> [MSG:Sleeping] but does not actually do anything, everything is still powered including the motors.

chamnit replied and said:

> Grbl's sleep command just disables the stepper enable pin, based on your $4 step enable invert setting. It's not tied to your stepper driver sleep pin.

and then he closed the issue.

What I was going to ask next was:

> So do I need to connect the Enable pin on the Easydriver to the Arduino.
> Because I've just made $4=1 and it still does'nt work.

> On the Easydriver the Enable pin is this:
> "ENABLE : This normally low input signal will disable all outputs when pulled high."

Any chance someone might now the answer.

chamnit, 2017-09-30:
People can still answer and help you. It's closed because it's not a bug. It's a user implementation problem. I tested the $SLP functionality myself shortly after your first message. It was operating normally. I close these regularly. Otherwise the issues threads get quickly out of control.

closed 2017-09-30 by chamnit

StuartB4, 2017-09-30:
Oh ok. No problem.
I still cant get it to work though.

X3msnake, 2017-10-02:
Why do you want to put the drivers to sleep. Usually people want to keep
the drivers always active since disabling the drivers means you will most
likely lose steps and have to re-home.

What are you trying to accomplish? Automatic driver sleep after x minutes?

StuartB4, 2017-10-02:
I want to put it to sleep after the job has finished.

1. I could do it automatically by placing the command at the end of the gcode program. 
2. Have a custom button in LaserGRBL to click after the job has finished.
3. Have a wired in switch to press when the job has finished.

When I use the first 2 options I get a response saying that it's sleeping, but the motors are still powered and cant be moved freely.

I have'nt tried the wired in switch because I need to know if it will work and which pin on the Arduino to connect the SLP or ENABLE pin on the easy driver. Also how would GRBL send the command to the Easydriver.

X3msnake, 2017-10-02:
First thing what are you using 328p or a mega. If the first you probably
dont have any free pins left and even is you have the change would have to
be very small because there is little to none space left on the 328p.

Maybe you can use one of the pins from the homing switches eventually but
you must know how to write the code in c to do that implementation since
grbl is highly optimized and does not use any arduino libraries.

I had no knowlage of the slp command is this something native to grbl? If
so you can look for the code that it runs and use it to enable the z homing
pin.

You should try if the sleep works as you want before by feeding it with the
right state manually.

StuartB4, 2017-10-02:
I'm using a genuine ATmega328P Arduino Uno, as you can see in the image above, there are plenty of pins left as there are only 7 connections to it at the minute. 
I think $SLP is a new addition to GRBL 1.1 as I can't see it mentioned in earlier versions.

neilferreri, 2017-10-02:
@StuartB4 Are you connecting the enable pin to your stepper drivers? I'd start there.

StuartB4, 2017-10-02:
No because I don't know where to connect it. 
With the Easydriver it does'nt need to be connected to anything for the driver to work. it seems to be for an input signal. The same as the SLP pin on the driver.
I just wondered if it needs to be connected to the Arduino to make the $SLP command work.
But I' m not sure if GRBL is able to send a signal to it anyway, thats what I'm trying to find out. Because just sending the $SLP signal does'nt seem to work, the motors are still powered and can't be moved freely.

109JB, 2017-10-02:
On many drivers the enable pin does not need to be connected for the driver to work, but it does have to be connected if you want to be disable the drive.  The easy driver needs a high signal to the enable pin to disable the drives. If it isn't hooked up you can't disable the drive.

vMeph, 2017-10-02:
if you dont have the enable pin conected then will not work
looking at the  Easydriver it should be connect the enable of arduino to the enable of the driver
[image]

StuartB4, 2017-10-03:
Thanks 109JB and vMeph. 
Thats what I was thinking but just don't know which pin on the Arduino, or even if GRBL is capable of sending the signal out to the SLP or Enable pin.

neilferreri, 2017-10-03:
Use pin 8 on the Arduino and connect it to your SLP or enable pin on the driver. Depending on which you use, you can invert the signal from the arduino using your $4 setting.

StuartB4, 2017-10-03:
Ok neilferreri I will give it a try tomorrow, 3:30 A.M. here, just going to bed at the minute. 
Thanks.

StuartB4, 2017-10-03:
Before I connect to pin 8 on the Arduino I'd just like to know 3 things.

1. How will the Arduino know what to do on pin 8 when I send the $SLP to GRBL.
    Don't I need to define what pin 8 should be/do, or will GRBL 1.1 know.

2. As my machine is a 2 axis (X,Y) do I need to connect both SLP or ENABLE pins to 2 seperate pins
   on the Arduino, or can I connect say both SLP pins or both ENABLE pins together with a wire
   and connect that to Arduino pin 8 so they both disable at the same time.

3. Do I need to add any resistors or just a wire from driver to Arduino.

Thanks.

vMeph, 2017-10-03:
you need to connect the enable pin of the arduino to all drivers enable pin no need resistors enable pin have a internal pullup to know if state is high or low

i actually never used the sleep command but most like you put the command $SLP on end of a gcode files i think needs to be at end of the % of the gcode file, or you can send command on the command line of the software you using to command machine like UGS platform has exemple

StuartB4, 2017-10-03:
Thanks vMeph.
One more thing. Could I wire it like this using 2 diodes, so that if one driver board developed a fault it would not go straight to the other board. Or would that affect the signal through pin 8 on the Arduino.
[image]

neilferreri, 2017-10-03:
@StuartB4 The $SLP command in GRBL will toggle pin 8.  It seems like you have the option to connect to SLP or ENABLE on your EasyDriver.  I wouldn't worry about that board developing a catastrophic fault.

vMeph, 2017-10-03:
I dont think you can do that sense there will be drop voltages on the diodes

vMeph, 2017-10-03:
Arent those smds by the enable pin diodes?
I dont know those drivers

StuartB4, 2017-10-03:
Ok Thanks. No, they are resistors.

StuartB4, 2017-10-03:
I've connected both the X and Y axis driver Enable pin to pin 8 on the Arduino and it seems to be working.
Did'nt need to change $4.

Thanks very much for all the help, I really appreciate it..

X3msnake, 2017-10-03:
Cool ;)