gnea/grbl #285: modifying 30khz step rate limit (20mhz uno)

samitray, 2017-10-01:
currently running grbl compiled with FCPU 20000000 on 328p with 20mhz crystal.
grbl has the max rates capped to 30khz, however i could only locate commented 
out defines for MAX_STEP_RATE_HZ. There are DEFAULT_X_MAX_RATE 
defines but they are rate for individual axis.
If max step rate is to be set around 40k, what needs to be changed?

Cheers

109JB, 2017-10-01:
There is no default coded limit. The step limit is set by your maximum rate and steps/mm settings. If you have changed the crystal then you should be able to increase your maximum rate without any code changes. Only changes to the $ settings should be needed.

chamnit, 2017-10-01:
The 30kHz is a primarily a function of how quickly the AVR processor can service a step interrupts and the amount of processing Grbl needs to do when under maximum streaming conditions.. Grbl does not actively check for the 30kHz step limit. It's a recommended limit that I've tested to cover most foreseeable circumstances.

closed 2017-10-01 by chamnit