gnea/grbl #286: GRBL does not connect with any softwares

KyuuKun, 2017-10-01:
I'm using grbl to construct a CNC Designer that I baptized of Daccin. My problem is that when I connect the arduino with any software (Like Universal GCode Sender) sometimes it connects but sometimes no.
How can I fix it?
Another question, is there any method to use GRBL with NodeMCU and its internet connection?

Sorry for my bad english.

X3msnake, 2017-10-02:
Are you using a cloned arduino with the ch340 serial converter ic?

This chip is known to be the cause of a lot of drop connections and other
com problems...

One that works is the cp2102

KyuuKun, 2017-10-02:
How can I check this?

vMeph, 2017-10-02:
[image]

109JB, 2017-10-02:
I have many clones with the CH340 usb chip. While they do have communication problems, I have never had one that would not connect or stay connected. The problems I have had are that with fast throughput, the chip can drop some characters from the serial stream which can be a problem for CNC. However they always connect, and always work except for the problem mentioned above. I still use the CH340 clones for Grbl testing all the time.  Also, the problems I had were with the CH340G chip, the newer CH340T chips may or may not have the problem. Only testing would flush that out though. I think the problem is elsewhere.

What kind of USB cable do you have? Does it have a ferrite bead to reduce noise on the line? Does the USB wire run near other wires that could be introducing noise?

biasedlogic, 2017-10-02:
Another thought: does your computer have a Bluetooth connection? I used to have erratic problems on my laptop until I found out, that there were virtual Bluetooth serial ports with com port numbers assigned, and for some reason my USB serial cable was getting the same com numbers. The Bluetooth software was periodically querying its ports, thus opening the serial port of my adapter and blocking it from use by other software.
This scenario happened on more than one laptop in the past, so it's not an isolated case...

zeevy, 2017-10-04:
I had similar issue, got rectified by changing the USB cable.
Try changing the USB cable. (preferably with one with ferrite bead)

alandroid, 2017-10-07:
[image]

Is the above issue in the screenshot related? Do i need to get an official Arduino board?

gerritv, 2017-10-07:
You need a newer version of Gcode Sender, 2015 is way before Grbl 1.1 came out. The protocol changed amongst many other improvements.

alandroid, 2017-10-07:
I downloaded a newer version of UGS and now i am able to connect but it appears that i keep getting the following output repeatedly in log, over and over, how come? Doesn't seem right

`
Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok
<Idle|MPos:0.000,0.000,0.000|FS:þ`

alandroid, 2017-10-07:
I got UGS 1.0.9 and was able to connect but am getting the following output in console repeating:

`Grbl 1.1f ['$' for help]`

And also i am unable to move my motors but that may be stepper drivers not tuned yet..

closed 2017-10-10 by chamnit

X3msnake, 2017-10-10:
You can Try bCNC too

blokkenmans, 2022-12-18:
I
 get the message No serial ports found

JBSchueler, 2022-12-19:
> I get the message No serial ports found

Is the serial port visible in the device drivers of windows?