gnea/grbl #288: Door pin not honored during homing

devotip, 2017-10-10:
While homing is in progress the door pin has no effect.

Homing should be aborted or at least movements should stop as in hold

chamnit, 2017-10-10:
It does. Or at least it should honor it, because it's in the code. Do you have safety door enabled? Please share your $I build info string. It should show that you have it enabled.

devotip, 2017-10-11:
Here it is, during normal operation door pin is working, the issue is only during homing 
>>> $I
[VER:1.1f.20170131:]
[OPT:V,15,128]
ok

chamnit, 2017-10-16:
@devotip : Looks like I forgot to add the safety door compile-time option to be reported in the $I build info, so I can't tell if you have it enabled or not. When you trigger a safety door, what is the state Grbl is reporting? HOLD or DOOR?

devotip, 2017-10-17:
Afterall it was built with door option disabled

closed 2017-10-17 by devotip