gnea/grbl #289: Modal group violation when using M7 and M8 on the same line

cheton, 2017-10-10:
Using M7 (Mist coolant) and M8 (Flood coolant) on the same line will report a modal group violation error. I used to workaround this issue with separate lines, but I think Grbl should allow M7 and M8 on the same line.

```
M7
ok
M8
ok
M7 M8
error:21 (Modal group violation)
```

chamnit, 2017-10-10:
@cheton : Is this fact or preference? If fact, then please show me where it states this in literature. Otherwise, the LinuxCNC gcode standard states that any two commands in the same modal group in the same line is not valid. This includes M7 and M8 together. While M7 and M8 are allowed to be active at the same time, the modal group requirement should still be used. There isn't an exception for this rule that I'm aware of.

closed 2017-10-10 by chamnit

cri-s, 2017-10-10:
I know that there is a exception for M7 and M8 that it could stay on
the same line from
the group rules.

The Nist RS274NGC interpreter where linuxcnc is based have this modal
group table
where the exception is noted and i paste it below, taken from here:
http://ws680.nist.gov/publication/get_pdf.cfm?pub_id=823374
Now i'm not searching further the exception in literature for you, but
both linuxcnc and
Nist and any other commercial controller that i know have implemented
this exception and
most have documented this exception as some footnote.

Regards
Cri.

The modal groups for G codes are:
group 1 = {G0, G1, G2, G3, G38.2, G80, G81, G82, G83, G84, G85, G86,
G87, G88, G89} motion
group 2 = {G17, G18, G19} plane selection
group 3 = {G90, G91} distance mode
group 5 = {G93, G94} feed rate mode
group 6 = {G20, G21} units
group 7 = {G40, G41, G42} cutter radius compensation
group 8 = {G43, G49} tool length offset
group 10 = {G98, G99} return mode in canned cycles
group 12 = {G54, G55, G56, G57, G58, G59, G59.1, G59.2, G59.3}
coordinate system selection
group 13 = {G61, G61.1, G64} path control mode
The modal groups for M codes are:
group 4 = {M0, M1, M2, M30, M60} stopping
group 6 = {M6} tool change
group 7 = {M3, M4, M5} spindle turning
group 8 = {M7, M8, M9} coolant (special case: M7 and M8 may be active
at the same time)
group 9 = {M48, M49} enable/disable feed and speed override switches
In addition to the above modal groups, there is a group for non-modal G codes:
group 0 = {G4, G10, G28, G30, G53, G92, G92.1, G92.2, G92.3}

cheton, 2017-10-10:
http://linuxcnc.org/docs/2.4/html/gcode_overview.html#cap:Modal-Groups

It states a special case that M7 and M8 may be active at the same time, but I'm not sure whether it's allowed to be in the same line for this special case.

The reason why I asked this question is that I'm currently implementing a tool change feature in CNCjs, which will save and restore work position and modal groups when running a tool change macro. As M7 and M8 may be both present with $G (i.e. `[GC:G0 G54 G17 G21 G90 G94 M5 M7 M8 T0 F0 S0]`), I would expect I can run **"G54 G17 G21 G90 G94 M5 M7 M8"** in the same line to restore modal groups before resuming program execution.

KurtJacobson, 2017-10-10:
They can be active at the same time, yes. but they can not be issued on the same line. At least I don't know of a control on which they can be. On LinuxCNC issuing M7 and M8 on the same line gives the error: `Two m codes used from same modal group`

chamnit, 2017-10-10:
@cheton : I can't help you unless you can show me explicitly (page number or link) where it shows  issuing `M7 M8` together on the same line is valid. I've read both the NIST and LinuxCNC documentation thoroughly. I'm fairly certain that I'm right.