gnea/grbl #292: Losing steps / gaining steps??

brakthehun, 2017-10-17:
[image]

So,, I built a 50 watt CNC CO2 laser and it started doing this a few months ago. I can't figure out why. I tell it to do 3 passes and the lines slowly get off in both the X and Y. Does anyone know why this would suddenly happen??

It will wander in both positive and negative.
It Does this on any Gcode I use.

Arduino UNO with a screw terminal shield.
3 TB6600 stepper drivers driving 84 ounce/inch nema17s at 1.8 amps. 
2 steppers running Y,, 1 stepper running X. 20 tooth gT2 on a belt drive
80 steps/mm
does this at every speed. It worked fine before at 15,000 mm/minute (Y at 10,000)
does this at every acceleration. Worked fine at 1200 mm/s/s (Y t 800)

X3msnake, 2017-10-17:
mechanical backlash?

brakthehun, 2017-10-17:
I can't imagine both axis developing mechanical backlash while losing / gaining steps randomly.
If it was mechanical backlash,, what would you do to fix / compensate??

X3msnake, 2017-10-17:
You are right probably not backlash,

Maybe dropping packets in the serial? are you using a clone arduino with
the ch340 chip.
EMF interference? Arduino board near any motors or power supplies?
Bad power to the motors? Recheck all motor connections to the motor and to
the drivers

Did it ever worked right? What was the last change you made before you
started noticing this?

btw why are the traces all shaky? is it supposed to be that way

brakthehun, 2017-10-17:
I am using an Arduino branded board with the 16U2 serial chip.
It did work correctly for a few months.
The traces are shaky because I'm running it 8,000 mm/minute so it doesn't fully cut through the cardboard and will leave a small line.
The board is somewhat Isolated. It is kinda near the 25,000 volt laser power line,, but the insulation is rated for 40,000 volts. It worked originally so I'm not sure if that would affect it. I can wrap the cable with some aluminum tape or something??
The connections seem to be good.

X3msnake, 2017-10-17:
Can you please post your grbl settings

What changed in your config? Added any new electronics changed something in
the machine the controller the pc?

What arr you using as a sender

brakthehun, 2017-10-17:
The config has Z homing disabled.

I have another laser (2.5 watt blue diode) about 4 feet away. I plugged that computer in to the CO2 laser and it behaves the same.

I am using laserGRBL.
I tried Gcode created in LaserGRBL and Gcode created in Inkscape and it does the same thing.
The smooth curves are created in Inkscape. The rough ones are from LaserGRBL. The vector program literally follows the bits on the bitmap.

I will send you the machine settings tomorrow.
I added an axle to join the two Y motors on the idler pulley side of the belt.
I added a focuser tray and swapped out the 5" lens with a 3" lens. That's about it for the major things.

chamnit, 2017-10-17:
I would check your mechanical components. Particularly your stepper pulleys. Make sure the set screws are seated and tight.

brakthehun, 2017-10-17:
Screws are tight. 
[image]

chamnit, 2017-10-17:
@brakthehun : Your settings look reasonable and normal. If you are absolutely sure that your CNC has no mechanical backlash (easy to check with a simple gauge and manual motions), then it's likely electrical. There have been some instances in the past where electrical noise would interfere with Grbl's step pulse and direction outputs. It looked similar to mechanical backlash. Given that you have a high voltage laser power line near your controller, it would be a likely source. 

BTW, I'm pretty sure that wire insulation doesn't block EMF. It only inhibits current and arcing. You'd need a Faraday cage or something similar to keep the EMF from inducing current in nearby circuits.

brakthehun, 2017-10-17:
Interesting. That will be my next route then.

X3msnake, 2017-10-18:
Cool keep us posted. ;)

indyucoder, 2017-10-18:
Ciao, have You checked the current limite trim offline the stepper drivers?

brakthehun, 2017-10-18:
The stepper drivers are all giving 1.8 amps.
I set $1=255 so the steppers would keep hold.
I tried shielding the Arduino and stepper drivers from the Laser power supply.
I replaced the Arduino.
Still goofing up.

vMeph, 2017-10-19:
you have what single HY-DIV268N TB6600 ?
one off the problems is most like your TB6600 microsteps switch table chinese is wrong
chinese TB6600 have the wrong microsteps table on drivers from what i see you have that you have that running AT 1/16 microsteps so you have switches ON OFF OFF?
if so try with OFF ON ON and let me know results

brakthehun, 2017-10-19:
The movement is nearly correct. I tell it to move 300mm and it moves 300mm +/- the jitter. 
I don't think this would account for losing / gaining steps.
I can post a pic of the set up.

vMeph, 2017-10-19:
isnt this you got?
http://lghttp.57222.nexcesscdn.net/803B362/magento/media/catalog/product/cache/1/image/650x/040ec09b1e35df139433887a97daa66f/t/b/tb6600_2.jpg

if so you runing at most like at 1/8 ON OFF OFF

brakthehun, 2017-10-19:
https://smile.amazon.com/gp/product/B016ZJS1FA/ref=oh_aui_search_detailpage?ie=UTF8&psc=1

brakthehun, 2017-10-19:
[image]
[image]

swarfer, 2017-10-19:
no one has mentioned step timing yet.
most drivers need some minimum delay between a direction change and the next step pulse.
GRBL does not have a specific setting for this IIRC.

all those lovely unshielded cables will affect the perceived delay in that they can affect rise time.
and keep the power wires (from PSU, and to steppers) *far away* from step/direction wires!

brakthehun, 2017-10-19:
It looks like I'm headed that direction - rewiring everything. It is kinda nice to do that though because everything always ends up looking better. Even when I run it slowly,, it still does the same thing.

X3msnake, 2017-10-20:
Does anyone know a good source of information on how to shield stuff
against EMI?

biasedlogic, 2017-10-20:
There's a ton, but most stuff starts with 'take a black cockatrice and a pint of fresh blood to the haunted crossroads when at full moon...' and gets rather gross soon after...
Let's start with grounding - try drawing up how your ground is connected and how the power lines are laid out

X3msnake, 2017-10-20:
@biasedlogic

I don't have a machine design yet, just looking to learn more, it's kind of
hard to know what is good information when you are starting on a subject
hence my question :) "Grabs a chicken"

jahnj0584, 2017-10-21:
Set the pulse delay to the default of 10. Use 1/4 uStepping. Unshielded
wire is not the concern here. I run 7000mm/min in my heavy machine and hold
0.01mm all day. I used shielded wire for my steppers but the drain wire
isn't attached so it's effectively doing nothing. My spindle IS grounded at
the VFD end.

No noise issues. No step or accuracy issues.

My first two machines had 0 shielding and performed the same as this one,
from the stepper+driver side of things.

JBSchueler, 2017-10-21:
I noticed you are driveing the steppers with 2.5A while, as you mentioned, the steppers are 1.8A.
This way, the steppers getting to much current and end up with losses within the coil.
[image]
dip 4-5-6 should be ON-OFF-ON (1.5A)

I had the same issue with my TB6600 stepper driver when I used to much current.

Another issue could be the power supply, can it deliver the 5A needed for the steppers?

jahnj0584, 2017-10-21:
The switches should be set to one increment above your motor amperage. That
is a setting for peak amperage, not a constant output.

JBSchueler, 2017-10-23:
@jahnj0584 : Correct, but I noticed on my driver it was losing steps when I set my drives to 1.5A on my 1.0A steppers. Switching to 1.0A solved my problem of losing steps.

brakthehun, 2017-10-25:
I fixed it last night. I tried many things including rerouting all the cables,, but what fixed it is moving the Arduino to an isolated area inside the laser enclosure and using a polulu shield for stepper drivers.

X3msnake, 2017-10-25:
Was it the tb drivers?

brakthehun, 2017-10-26:
I will test them soon. I hope not!!

langwadt, 2017-10-26:
found on the openbuilds site: "if people were to stop using tb6600, technical support posts would drop 75%" ;)

brakthehun, 2017-10-26:
I have a few machines running TB6600s. Good to know. Thanks.

chamnit, 2017-10-26:
Thanks for the report. I’ll close this issue now.

closed 2017-10-26 by chamnit

vMeph, 2017-10-26:
using a extra shield polulu  they done to use with other drivers, even know you cold adapt to other drives  seems to me like a waste sense you have external drives tb6600
your problem most like is EMI on lines, did you try use shielded cables with proper ground to drain the EMI?
https://www.youtube.com/watch?v=84udcsEC7zI