gnea/grbl #293: C axis

brakthehun, 2017-10-17:
It would be nice to use the Z axis as a C axis for plunge knife and saw machines. I have a contouring saw mill design in my head for creating snowboard cores. I just need a C axis for that.

chamnit, 2017-10-17:
@brakthehun : It certainly would, but if you haven't been able to tell, the Arduino Uno flash space and memory is maxed out, especially for certain critical configurations. Literally a few hundreds of bytes of flash left. Additional axes will be supported in the 32-bit ARM version.

closed 2017-10-17 by chamnit

brakthehun, 2017-10-17:
Which ARM chip are you going to standardize on??
Is there an active discussion somewhere??

X3msnake, 2017-10-18:
Sonny is working on a hardware abstraction layer (HAL) that make it easy to
use the firmware in most ucs

There is a thread here
https://github.com/gnea/grbl/issues/67

X3msnake, 2017-10-18:
List of known active port projects

https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#grbl-porting

There is also a official mega version
https://github.com/gnea/grbl-Mega

brakthehun, 2019-10-04:
I was thinking instead of a Z axis,, it would have a C axis. It would interpolate the C direction based on the movement of X and Y automatically. It would still be 3-axis.

Anakin-Chauhan, 2020-09-15:
Is there any we could use z axis driver for extruder motor that draws the filament?