gnea/grbl #294: CH340 and speed setting

swarfer, 2017-10-18:
Hi all

This is not an issue, just an idea and a data point.

Serial speeds are not exact multiples of the Arduino clock and I had this idea that maybe by selecting an exact multiple we can reduce or remove the problems with CH340 USB chips.

To this end I looked up the speeds available for the CH340.
I set mine to 128000 on config.h , flashed GRBL 1.1 and then ran a couple of big files through it.  First one was about 40 minutes of cut time, second was 1H33.    GUI was Candle.  USB cable is a decent one with ferrite filters.

128000 is exactly 1/125 of the 16mhz clock, whereas for all the common serial speeds there is some clock error.

Result?  no errors in comms so far.   I will continue to throw Gcode at this testbed (just a clone UNO lying on my desk) and see what happens.

X3msnake, 2017-10-18:
Important question. Did you get errors running with that board on other
speeds? I have read that some boards with the ch don't  cause problems...

chamnit, 2017-10-18:
@swarfer : I believe this was tested already, and there wasn't a noticeable improvement to reliability. At least at the time. The suspected problem with the CH chip is the RX/TX buffers would overflow and lose data without notification. This would occur if the communication bandwidth was close to saturated.

langwadt, 2017-10-18:
uart runs at 16x the baudrate so 128000 still has clock error; 16e6/(128000*16-1) = 7.8125

109JB, 2017-10-19:
I guess I was the one that opened this whole can of worms wrt the ch340 chip and Grbl.

I tried the same thing, adjusting the baud rate to match clock rates, and much much more. I kept testing for a long time. The baud rate did seem to have something to do with it at first, but I eventually decided that it was not part of it.  Much of my testing is described in this thread https://github.com/grbl/grbl/issues/845, but I did much more than this. It included testing at baud rates that I calculated to be good candidates for the 16 MHz arduino clock, which I can't say for sure but I would bet included 128000. All baud rates I tested resulted in errors.

Also when you say a big file that had a cut time of 40 minutes, that isn't a big file at all. I ran files that were millions of lines long and had run times approaching 20 HOURS.  Sometimes the errors would occur right away, but I had some where there were no errors for many hours.  When the errors would occur was totally unpredictable. During all of this testing I wrote programs for the arduino and sending programs specifically to flush out the problem. I used many means to sniff the data stream at the Arduino TX and RX pins including using an old laptop I had that had a real serial port, other arduinos (ch340 and 16u2 equipped) as sniffers, multiple compare programs, and the list goes on.  I would estimate that I invested about 500 man-hours investigating this issue with many more hours of the computer/Grbl running automated tests.

One caveat is that all of my testing was on CH340G chips and I see many clones now come with a CH340T chip.  I don't know what the difference is, but I did no testing on the CH340T.

After all of my documented testing I still did more and what I concluded was that the errors always coincided with a status report request. I surmised that the errors could be the result of the ch340 transmit buffer somehow interfering with the receive buffer, but the firmware for the CH340 isn't available to make a determination of what is really happening. All I could say in the end was that the CH340 doesn't work 100% reliably.  Even if it is 99.9999% reliable that isn't enough for me. For som it may be, but not for me.

Also, if the baud rate was simply not compatible with the Arduino clock, then why does it work flawlessly with a reflashed 16U2. If the arduino baud rate compatability were the culprit then it would not matter which USB chip was installed and the errors would occur on them all.

biasedlogic, 2017-10-19:
> Also, if the baud rate was simply not compatible with the Arduino clock, then why does it work flawlessly with a reflashed 16U2.

That would be, because the 16u2 would have the same clock error and so the baudrates would be identical, if both off.
But I'd also say there's another dog buried here

109JB, 2017-10-19:
> That would be, because the 16u2 would have the same clock error and so the baudrates would be identical, if both off.
> But I'd also say there's another dog buried here

The original poster said nothing about baud rate matching. He said

>Serial speeds are not exact multiples of the Arduino clock 

and never mentioned the frequency of the ch340.

However, I tried that in my testing too but I did take into account the differing crystal frequencies. The CH340 uses a 12 MHz crystal whereas the Arduino uses a 16Mhz.  I used baud rates that yield 0% baud rate error at BOTH crystal frequencies and it still didn't work. Incidentally 250000 is a baud rate that has 0% error at both 12Mhz and 16Mhz. I can guarantee that errors still happen at 250000 baud when using a ch340 because I know I checked that rate.

When doing my testing I used a baud rate calculator for deciding which ones to test.  I just went back to it and added 128k to the list and it is actually a horrible match as can be seen in the link below:

http://wormfood.net/avrbaudcalc.php?bitrate=300%2C600%2C1200%2C2400%2C4800%2C9600%2C14.4k%2C19.2k%2C28.8k%2C38.4k%2C57.6k%2C76.8k%2C115.2k%2C128k%2C230.4k%2C250k%2C.5m%2C1m&clock=12%2C16%2C+20&databits=8

109JB, 2017-10-19:
@swarfer When you said:
> Result? no errors in comms so far. 
How did you test that?  The reason I ask is because the CH340 errors I discovered may not be picked up by Gbrl. In order to insure that there are no errors, you either have to monitor the serial port at the Arduino TX/RX pins using a separate program, or you have to turn on Grbl echoes and compare character by character that each line sent was received by grbl. The CH340 errors are insidious and can still result in corrupted values being sent to grbl and looking like valid G-code to grbl, where a Grbl error isn't thrown. For example, the following g-code lines:

G1 X2.588550 Y3.**482510 Z-0.100000 
G1 X4.644990 Y5.4825**10 Z-1.100000 
G1 X6.701420 Y6.482510 Z-2.100000 

could become

G1 X2.588550 Y3.10 Z-1.100000 
G1 X6.701420 Y6.482510 Z-2.100000 

where the bold part in the original is simply dropped due to the ch340 errors.  What Grbl would see in this case is still valid G-code, but the coordinates and sequencing of moves is drastically different that the original intent. What gets dropped by the ch340 is random and could cause grbl to throw an ertor or it might not.  My testing confirmed that the above scenario can and does happen.

X3msnake, 2017-10-19:
> Forgot to mention that this isn't the only type of error that can be
> encountered but is an example of one that would not result in grbl sensing
> an error.
>
> Incidentally, In my GUI I have implemented a simple error detection scheme
> that uses the Grbl echo function. When a command is sent to Grbl, and a
> Grbl echo received at the GUI, the sent line is compared to the echo to see
> if they match. If they do then streaming continues, if they don't match the
> GUI immediately stops streaming and issues a soft reset to halt operation.
> In order to do the compare the sent line is converted to all capital
> letters and all whitespace is removed to get an apples to apples
> comparison. This is exactly the same as what Grbl does before returning the
> echo. I have elected to leave echoes turned on at all times. There is a bit
> if a performance penalty for doing this, but for me it is a tradeoff I am
> willing to make to insure accurate data transmission.

@109JB
what are you using as sender?

109JB, 2017-10-19:
I use a sender I wrote myself using Visual Basic.  I had made my previous one for V0.9 available but this new one is V1.1 compatible and has many more features but is not ready for public consumption. I have not had much time to work on any personal projects since my mother passed away last December.  I have had to deal with her estate, my ailing father, and the multiple properties they owned.  It has taken just about all my spare time and what I do have i spend on my own family. If I ever do get time to finish it I will make it available but it will likely be some time before that happens.

X3msnake, 2017-10-19:
Ok cool best wishes with your personals.

What is missing for it to be public worthy?

109JB, 2017-10-19:
It needs mostly testing, but there are features I added or changed that I have to flush out. 

1. I added automated probing routines for things like bore probing, corner probing, edge probing, etc. Needs testing and adding user settings for things like probe distance, tip diameter,....

2. I changed the way my canned cycle routines for G81-89 work and need to test them to make sure they are still working right.

3. I added the echo compare I described before, but want to put in an option so the user can opt in or out of this checking.

4. Reworked things for V1.1 support and need to do testing to verify all is good.

Another thing I want to try is using a second Arduino on a second com port for things like a MPG wheel.  This would work by having the second Arduino connected to MPG wheel \, buttons, etc and send signals to the GUI which would be interpreted by the gui and sent to the Arduino with Gbrl.  Here is a shot of what it looks like now:

[image]

Extremator1, 2017-10-20:
@109JB 
What do you think, is it possible to avoid these errors by the bypass of CH340? 
That is, the connection chain will: 
(PC sender) - (PC COM port, if there is) - (converter RS232-to-UART, like MAX232 chip) -
- (GRBL's Arduino UART)

109JB, 2017-10-20:
The problem is in the CH340, so eliminating the ch340 from the chain would work to get rid of these problems that I discovered.  In fact, I did get a MAX232 and did some testing like what you are describing, and did not see any problems. However, I only did a limited amount of testing and ultimately decided that it was just easier to get an Arduino with a 16U2 converter chip since the problems with that chip have been eliminated with upgraded firmware.  I have not had any problem with the 16U2 and upgraded firmware ever.  I bough my last Arduino Clone with 16U2 for $4 including shipping to my door.  I would recommend this as the easiest solution.

Extremator1, 2017-10-21:
@109JB 
Thank you for deep research this topic!

X3msnake, 2017-10-22:
@109JB <https://github.com/109jb>

Did you ever published any posts beside the ones here on the git about this
subject, I would like to compile this info to put up on the GRBL wiki

I think this is important subject to be featured on the wiki since it is
not already.

Extremator1, 2017-10-22:
Sure!
See wiki/Known Issues
https://github.com/gnea/grbl/wiki/Known-Issues

X3msnake, 2017-10-22:
Ok Cool

109JB, 2017-10-22:
The issue is already published in the WIKI

https://github.com/gnea/grbl/wiki/Known-Issues#usb-to-serial-transmission-errors

X3msnake, 2017-10-22:
> The issue is already published in the WIKI
>
> https://github.com/gnea/grbl/wiki/Known-Issues#usb-to-
> serial-transmission-errors

Yes you are right, i missed it, sorry. Thanks

chamnit, 2017-10-23:
Closing since this is an issue retread.

closed 2017-10-23 by chamnit