gnea/grbl #295: alarm 1 state

JamieAKing, 2017-10-21:
every time I enable the hard limit switches and enter a move command grbl locks up. initially I thought that this was a limit switch problem but if I show the verbose window and leave grbl in the alarm state on startup I can see that the limit switches are operating properly. i.e. when one is pressed the corresponding pn-x,y,z,p is displayed
[image]

you can see where I depressed and released the x limit switch here

JamieAKing, 2017-10-21:
this is what happens when a move command is issued
[image]
as you can see it is not a limit switch triggering the alarm

109JB, 2017-10-21:
It still could be a limit switch trigger. The status report shows the state at the instant that it is given.  If a limit is triggered between status reports, but is untriggered before the next one it will throw the alarm, but not show on the next status report.  It could be noise that is triggering the limit which could be of such a short duration that it doesn't show on the reports.

biasedlogic, 2017-10-21:
I had this with a plotter - the end stops were triggered the instant a motor would get power. Noise. RC filter was enough to kill the problem.

JamieAKing, 2017-10-21:
Hmm I have the limit switch wired in the nc configuration. So that when the switch is operated the connection is broken. It was my understanding that this way the false triggering was less of an issue but as I write this I realise that the ardunio pins are in a low state when connected to ground and then go high when the switch gets touched and breaks the circuit. It is obvious now that it should be the other way round. If the pin is constantly high then an induced voltage from a motor will have no effect. 
God damn those switches were so hard to connect too. 🙈 I have read else where that capacitors can help. I think I will try that before rewiring every thing

biasedlogic, 2017-10-22:
>I have the limit switch wired in the nc configuration. So that when the switch is operated the connection is broken. It was my understanding that this way the false triggering was less of an issue

That's the best way to wire it, ignore my comment on noise then. Input shorted to ground is pretty much noise immune. 

>If the pin is constantly high then an induced voltage from a motor will have no effect.

That's not so, the internal pull ups are not so strong, so noise can easily trigger them low

JamieAKing, 2017-10-22:
So what is the solution to the pull up resistor not being strong enough. I read through the limit switch thread on this board and by the end of it in afraid to say i was no closer to a solution. Opto  isolators, no opto isolators new pcbs etc. My machine bed is large enough that I can just work in the middle for the time being but it would be nice to have that extra safety factor of the the limit switch. One extra 0 here or there could result in devastation!

biasedlogic, 2017-10-22:
The solution to a not so strong pull up is adding external pull up... Anything around 220 to 470 ohm would be perfect. Adding 100nF cap to ground is a good idea too

JamieAKing, 2017-10-22:
Ok so as you can probably already tell I’m so up on the electronics. I have done a few google searches and from what I can understand when using nc switches the 5v supplied from the ardunio pin is taken to ground.  Then when the switch opens the ardunio pull up resistor holds the pin high. As I am experiencing problems when the switch is closed how will increasing the pull-up help me? 
Sorry if this is stupid question but there is only so much googling that can be done before interest levels start to drop. I would much rather be in the garage using the fancy new cnc than trying to learn the dark arts of electron manipulation

vMeph, 2017-10-22:
https://github.com/gnea/grbl/wiki/Wiring-Limit-Switches

closed 2017-10-23 by chamnit

biasedlogic, 2017-10-23:
>As I am experiencing problems when the switch is closed how will increasing the pull-up help me?

It won't. Noise is not your problem. Or at least it doesn't seem to be a problem on the limit switches. My initial hypothesis was based on the assumption you are wired the default way (n.o.). Since it's not so:

>ignore my comment on noise then. Input shorted to ground is pretty much noise immune.