gnea/grbl #297: Soft limits dont work in negative workspace

jahnj0584, 2017-10-25:
Just noticed this today. My machine homes in the top left corner, away from me so when it homes my coordinates are -1200 X and 0 Y. This means soft limits do not function, because they are set to only trigger at +1200 X and +630Y while the machine actually moves the opposite way for both axis. This soft limit value should be derived from the homing position.

chamnit, 2017-10-25:
Soft limit is and should be derived from the homing position. Soft limit errors occur when machine position exceeds the machine volume. You are either looking at work position or have your coordinate frame misconfigured (must follow the right hand rule).

jahnj0584, 2017-10-25:
I assume this version of it?
http://www.cnczone.com/forums/attachment.php?attachmentid=25504&stc=1&thumb=1&d=1389507392

Am I not able to use soft limits then because I use the top left as my origin?

chamnit, 2017-10-25:
can you clarify what you mean by trigger at 1200 X and 630 Y and moves in the opposite way? 

From home, if you command `G53 G91 X1`, which way does the machine move? And what does the machine (MPos) coordinates, not WPos, say?

From home, if you command `G53 G91 Y-1`, which was does the machine move? And what does the machine (MPos) coordinates, not WPos, say?

jahnj0584, 2017-10-25:
powered on. Homed the machine. Machine position says -1190 and -4 (pull off
distance).

X1 moves my gantry 1191mm to the right, and Mpos read +1.00
Y-1 moves my gantry away from me by 3mm to an Mpos of -1.00

chamnit, 2017-10-25:
Did you command the `G53 G91` with them? It shouldn't have moved that way.

jahnj0584, 2017-10-25:
yes exactly as you wrote.

Here are my settings
$3=5 (Step direction invert)
$5=1(limit pins invert, bool)
$20=0 (soft limits) (until this is sorted)
$21=1 (hard limits)
$22= 1(homing cycle)
$23=1 (homing Dir Invert)

$24= 800(homing feed, mm/min)
$25= 1800 (homing seek, mm/min)
$26= 50(homing debounce, msec)
$27=3 (homing pull-off, mm)
$30=24200 (Max spindle RPM)
$31=0 (Min Spindle RPM)

$110= 6200x rate, mm/min)
$111=6700(y max rate, mm/min)
$112=5200 max rate, mm/min)

$130=1194(x max travel, mm)
$131= 627(y max travel, mm)

My Z does not have a switch on it.

chamnit, 2017-10-25:
Ok, seems G53 G91 is broken. Or I did that intentionally. I remember thinking about this once and it may have been intentional. Need to look into whether G53 is strictly absolute machine coordinates or if it allows incremental. 

Remove the G53 and do it again for me. 

When does MPos say when you get a soft limit alarm? Can you show me the MPos, WPos, and the command you used to cause the error?

jahnj0584, 2017-10-26:
powered on. Homed the machine. Machine position says -1190 and -4 (pull off
distance).

G91 X1 moves my gantry 1mm to the right, and Mpos reads -1189
G91 Y-1 moves my gantry towards me by 1mm to an Mpos of -5

chamnit, 2017-10-26:
Ok. Can you show me what happens during a soft limit alarm. The initial MPos, WPos, and the command that throws the alarm?

109JB, 2017-10-26:
FYI, G53 should only work in absolute positioning, not in incremental mode.

chamnit, 2017-10-26:
@109JB : Thanks for the confirmation. I thought I did it right.

jahnj0584, 2017-10-27:
It seems like I maybe just needed a reset. I tried this without the motors
on, but  any jog that puts me outside the envelope gets an error 15 pop up.

jahnj0584, 2017-10-27:
I assume I cant go closer then 1mm to the home position because of the
potential triggering of the switch?