gnea/grbl #298: Trouble flashing 1.1

francis110, 2017-10-26:
I have flashed 0.9 with no issue, but when I wiped the arduino and sent 1.1 via adding library and the GRBL Upload example method and HexUploader methods it seems not to work.

When I click on Serial Monitor a backwards question mark appears and does not display "Grbl X.Xx ['$' for help]"

Something worth noting is when I use the GRBL upload example method it displays this error:

Sketch uses 30508 bytes (94%) of program storage space. Maximum is 32256 bytes.
Global variables use 1633 bytes (79%) of dynamic memory, leaving 415 bytes for local variables. Maximum is 2048 bytes.
Low memory available, stability problems may occur.

brakthehun, 2017-11-04:
How do you "wipe" the Arduino??
What baud rate are you using??
What program are you using to attempt to talk the the Arduino??

brakthehun, 2017-11-04:
I don't think Serial monitor can properly connect to GRBL.

X3msnake, 2017-11-04:
Are you connecting at the right baud rate?

1.1 uses 115200

francis110, 2017-11-04:
baud rate was the issue. Sorry, I am an arduido n00b. Thank you!

closed 2017-11-04 by francis110

X3msnake, 2017-11-04:
No stress ;)