gnea/grbl #299: What is the expected behavior of spindle enable in Laser Mode?

Rowbotronics, 2017-10-26:
With Laser Mode enabled and USE_SPINDLE_DIR_AS_ENABLE_PIN set, is the spindle enable pin supposed to toggle when switching between G0/G1 movement modes, or will it only turn off when the Laser is disabled completely with an M5? Our Laser needs to be run with a constant PWM and turned on and off using the enable pin in order to respond quickly enough at engraving speeds.

chamnit, 2017-10-26:
I think I remember installing it so the PWM remains on until changed. The enable pin regulates the off and on. There might be a compile time option that alter its behavior too. It’s simple to test if you have an Arduino and multimeter

closed 2017-10-26 by chamnit

Rowbotronics, 2017-10-27:
Thanks for the quick response.  I checked it out with a multimeter and oscilloscope, and when doing a G0 move it turns off the PWM output and leaves the enable pin high. Enable turns off when spindle is disabled using M5.

You said there might be a compile time option, would that be in config.h or somewhere else? The only thing related that I saw was SPINDLE_ENABLE_OFF_WITH_ZERO_SPEED, which isn't quite what I need.