gnea/grbl #300: Slow start / spindle ramp-up

xsrf, 2017-11-04:
Hi,
every time I enable my spindle with max rpm, my psu voltage drops drastically, because the spindle initially draws a huge current. This causes problems and sometimes the controller resets.
Also, the high inrush current is bad for the connectors and motor brushes.
It would be a nice feature, if grbl supported a ramp-up setting where the spindle would actually be slowly ramped up/down to the speed set via S command.
Guess this would be helpful for lots of cheap boards with simple mosfet spindle controller without extra inrush current protection. What do you think?

biasedlogic, 2017-11-04:
It's been mentioned quite a few times already.
In general, the start of the spindle is the business of a spindle driver. If the spindle driver is too dumb to run a soft start, you can implement the soft start with gcode, increasing set speed in steps and inserting delay in between

winder, 2017-11-06:
> Noted but not going to happen on the 328p due to flash and RAM limitations.

UGS has an optional feature to insert a dwell after the spindle start
command, I could add something to ramp up the speed as well if that's
useful. Create a ticket in the UGS github if you want to discuss something
like that.

xsrf, 2017-11-06:
@winder thx for the offer :) 
I may actually go for a hardware solution in my case (a simple current limiting circuit) ... I don't like the idea of a ramp-up made with dozens lines of g-code, added by a post-processor etc., because that would limit manual control drastically. You wouldn't be able to just turn the spindle on in any control software.
I really think, if not in hardware, the actual controller firmware should be the right place to do that...

@chamnit 
I'm new to grbl and I don't know how much space and ram the code actually consumes. If it's not possible to add the feature, well, then I won't complain. It's still a nice piece of software ;)