gnea/grbl #301: Hi All... I think the most updated version of Lasergrbl is 1.1 but I  saw a version n2.8 on SOFTPEDIA.com? Can anyone confirm what this might be?

fasteddyD, 2017-11-06:


arkypita, 2017-11-06:
Grbl and LaserGRBL are two different products. They are related but they are different.

The most update version of grbl is v1.1f
The most update version of LaserGRBL is v2.8.11

Grbl run on arduino (the electronic controller board you have on your engraver) and its work is to receive and execute the GCode commands translating them into engine motions.

LaserGRBL run on PC and its job is to send GCode commands to the controller board.
These are the basics.

closed 2017-11-06 by chamnit

fasteddyD, 2017-11-07:
Interesting.  So I can try to use LaserGRBL 2.8.11 to run my laser diode?  I am currently using vs 0.9, so I assume that I am using grbl rather than Lasergrbl.  Sorry for the stupid question but this software handles photos very well and is better than anything I have attempted to use up to now.  My only problem is that the x axis wanders during the burn, which offsets the photo, especially a larger burn.  I was looking at upgrading to vs 1.1 before I saw the version 2.8, which confused me.  I have downloaded 2.8.11 but not yet attempted to run it with the laser but it opens and looks just the same as what I am using.  I will try this one tonight and see what happens.  I really appreciate what you guys have done with this software and will be certain to hit the donation button as soon as I get this thing singing.  I bought the laser primarily for photos and even with the drifting problem your software does the job, so just need to figure this one out.  Thanks again... Ed

biasedlogic, 2017-11-07:
@fasteddyD : @arkypita descried it to you already very clearly, to quote:

**Grbl** is a software that run **on arduino** (the electronic controller board you have on your engraver). Its work is to receive and execute the GCode commands translating them into engine motions. Grbl does the hard work, without Grbl your marker would not move by one mm.

**LaserGRBL** run **on PC** and its job is to send GCode commands to the controller board (running GRBL [addition mine]).

These are the basics. There are a lot of program like LaserGRBL that can talk with Grbl, here is a list https://github.com/gnea/grbl/wiki/Using-Grbl and you can chose the one you prefer.