gnea/grbl #302: Buffer size is 128 or 127?

arkypita, 2017-11-06:
[Somewere](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#streaming-protocol-simple-send-response-recommended) I can read that buffer size is 127

> Grbl essentially has two buffers between the execution of steps and the host PC interface. One of them is the serial receive buffer. This briefly stores up to 127 characters

[In the same document](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#streaming-protocol-character-counting-recommended-with-reservation) I can read that buffer size is 128

> We know Grbl has a 128 character serial receive buffer, and the host PC can send up to 128 characters without overflowing the buffer

With $10=2 I can activate the report of Buffer State, and it show a value of 128

> Buffer State:
Bf:15,128. The first value is the number of available blocks in the planner buffer and the second is number of available bytes in the serial RX buffer.

What is definitely the buffer size?
It was changed from v0.9 and v1.1?

chamnit, 2017-11-06:
v1.1 was increased to 128.

closed 2017-11-06 by chamnit

arkypita, 2017-11-06:
Superb! I fix documentation inconsistency in gnea wiki.