gnea/grbl #303: grbl config

bwg16, 2017-11-07:
I am very new to cnc so please bear with me. I just got my cnc built and the electronics set up and everything is communicating just fine (at least I think it is).
So I thought I would at first use a pen to draw a picture of a basic shape produced through Inkscape and gcodetools.
Then I had ugs send the gcode to grbl through a cncsheild connected to arduino.
It produced a picture but it was downsized from about 6 x 6 inches to about a half inch square.
I spent the afternoon trying to track down where the problem is but am having no luck. I thought it might have something to do with my firmware settings but I don't know.
Can anyone recognize where I have erred?
Thks
Bruce

neilferreri, 2017-11-07:
@bwg16 If you jog the machine, does it move the correct distance on each axis?

chamnit, 2017-11-07:
Grbl's distance units are millimeters. Make sure that you have configured Grbl correctly. Also, check your g-code to see if it sets inches or millimeters mode (G20 / G21) depending on what it was generated for.

bwg16, 2017-11-07:
At first only x and y moved correctly.  Z was backwards.  I fiddled with $3 and it worked it i set it to $3=4. Then all three moved in the right direction

bwg16, 2017-11-07:
g20 or g21 seem to have the same results

neilferreri, 2017-11-07:
Is $13=1?

jahnj0584, 2017-11-07:
youll want to use this to make your parts on size
https://nathan7.eu/stuff/RepRapCalculator/RepRapCalculator.html#MotorStuffPrecise

bwg16, 2017-11-07:
No. $13 is 0

neilferreri, 2017-11-07:
Can you post or share a link to your Gcode?

bwg16, 2017-11-07:
[spiral_0003.txt](https://github.com/gnea/grbl/files/1448310/spiral_0003.txt)

neilferreri, 2017-11-07:
Ok, you're gCode's extents are less than 10mm square. The G21 at the start
of your code specifies that all units are millimeters.
That said, if you change that to a G20, preview your code before you send
it. Didn't look like a pen plotter code to me.

bwg16, 2017-11-07:
Thanks but I need to ask.  Are those gcode extents set by inkscape or gcodetools or ugs.  I don't wnat you to do all the work but if I could just get a start, I'll take it from there.

neilferreri, 2017-11-07:
That's coming from inkscape/gcodetools. I haven't used gcodetools, but I've
checked it out. I believe there's a scale that shows up on the page when
you run it. Your design might have been too small for you to see the scale.

bwg16, 2017-11-07:
Thanks to you I have a starting point.  Gcodetools operates kind of crazily (to me, anyway) but I'll go back and work from there.  Thanks again.

neilferreri, 2017-11-07:
No problem.  If you're just getting started, you might want to look at Easel
<http://easel.inventables.com/>from Inventables.  It is super easy to
generate basic gcode.  Their forums are a great place to get help too.

bwg16, 2017-11-07:
I looked at that but didnt find a driver for linux

neilferreri, 2017-11-07:
The driver is only necessary to communicate with the machine.  You can
export the gCode and use whatever sender you want.  I recommend CNCjs as
its UI is the most intuitive..

bwg16, 2017-11-07:
that makes sense . if i can get it to export gcode i can use ugs to send it to grbl

neilferreri, 2017-11-07:
Correct

bwg16, 2017-11-07:
thanks again neil.  you have been a BIG help.

closed 2017-11-07 by chamnit

biasedlogic, 2017-11-07:
I found using either Grbl-plotter or f-engrave (both free software) to get plotter-like g code based on svg picture (f-engrave can also do bitmaps) and Candle to preview it/send to machine to be easiest debugging combo, as each of these programs shows live how big the expected / generated code in real world units will be. 

Gcodetools i found to be very temperamental and it's got it's own scaling settings which are completely independent of inkscape dimension units.