gnea/grbl #304: SpinDir. pin goes wild during boot and again on USB connection.

joshk03, 2017-11-07:
I am using the very latest GRBL 1.1 on an Uno with a Protoneer v3.51 board.  I am using both SpinEn and SpinDir for my project.  The SpinEn pin works as expected.  However the SpinDir pin pulses my relay 3 times when I plug in the 12v power to the Arduino...  Then Another 3 times when I plug in the USB cable to the PC!  My SpinDir relay is running an inductive load, so the pulsing of the SpinDir pin needs to be solved before I finish the build.  Is this a pure bug, or is there any explanation that prevents it from being fixed?

chamnit, 2017-11-07:
This is the Arduino bootloader toggling pin D13. There is a FAQ answer written about this.

closed 2017-11-07 by chamnit

joshk03, 2017-11-07:
Ok, Thanks for the 'direction' chamnit. :)