gnea/grbl #305: Recover from last command guidelines (UI development)

arkypita, 2017-11-09:
Hi all. I am trying to develop a function to help the user to recover an interrupted job (hang, reset, disconnection) continuing from the very last command "fully executed" from Grbl when streaming a G-Code file.

The info I know are the last command that receive a "OK/ERROR" response and the last command I have send through the COM.

The operation I use to do for resume are:
- ask the user from where to restart (position) suggesting some significant options
- analyze the file until this position and compute last speed, power, and state of laser as well G90/G91 absolute and relative coordinates
- analyze the file and compute the correct final coordinates of the previous command

Then I:
- send a $H homing to fix uncertain position (optional) 
- send a fast move to that coordinate
- reapply speed, power, state and G90/91
- enqueue the file starting from that position

[www.youtube.com/watch?v=UwJAFp1pS-I]([image])

This work quite well, but I see that the "OK" response does not arrive when the command is "fully executed" but only when it is transferred to the planned buffer (i think). So the job doesn't start exatly from where left, but some step next.

I can obviously get back from some commands: the risk to double overwrite some parts is less annoying than left some missing part. 

However i ask: Is there a simple way to guess what is the very last "completely executed command"?

jahnj0584, 2017-11-09:
I usually go by the "lines sent" in the GUI and its always very close

arkypita, 2017-11-09:
@jahnj0584 do you mean the line sent counted from the GUI? it is very close but onward

jahnj0584, 2017-11-09:
If it says 2032 lines sent I'll start it from there. I run a mill and it
has yet to have to plunge violently ahead in the program.

It also depends how long your commands are. One line may take 250ms for the
machine to run while some may be 40 seconds.

winder, 2017-11-09:
I think you have to analyze the XYZ coordinate and Gcode to figure out which command was active. You can narrow it down to commands which have received an `ok`, but then need to verify.

@chamnit is there a deterministic way to decide how far back you need to search to find which `OK` command is responsible for the current location?

To analyze the gcode I suppose linear movement should be a matter of verifying that the current location is between endpoints.

Arc movements are more complicated if the arc is greater than 1/4 of a circle. I would probably start by checking the point-to-arc-center distance, then split the arc into line segments and apply the same linear movement check. There is probably a more precise way to find if a point is on a circles diameter, but you need to be sure your algorithm can tolerate rounding errors.

chamnit, 2017-11-09:
One non-ideal option: You can use the line number report, which requires uncommenting the USE_LINE_NUMBERS in config.h. Some OEMs use this to track what and when a line has been executed. But, it needs the GUI to send a unique line number `N` with each line. It also uses up a lot of memory (reduces planner buffer to make room) and communication bandwidth, neither of which are good for laser engraving (performance will suffer).

Another way is to observe and record the planner buffer size prior to aborting the job. It can be accurate if you know what's been 'ok'd by Grbl, but G2/3 arcs can make it undeterministic because it fills the planner buffer with small lines to generate the arc.

I'm sure there are other ways, but this has been one of things I've been wanting to improve in the ARM version, where I have the space to implement something better.

@arkypita : What you are asking for is something called 'program restart' in the CNC world. You have to be very careful about this, especially when dealing with a laser that can cause fires. It's very easy to incorrectly set the g-code modal state and the initial position at the resume point. A mistake like this will cause the machine to do something unpredictable and likely ruin the job or worse.

How a program restarts can also be machine dependent. You need to somehow move the machine into the correct position and set up its state (spindle, coolant, etc). You also have to deal with how to handle if there was a tool change mid-job, if you are supporting milling. Each machine is going to be slightly different and need custom routines to handle this type of stuff.

winder, 2017-11-09:
@chamnit it might be interesting to think about this problem in terms of what information GRBL would need to provide in order to implement 'program restart' without also implementing a gcode parser.

Even today you can trick GRBL into reporting the state at an arbitrary line of code by running the program in check mode and making a `$G` request after sending the lines to be skipped. You would still need to parse the program to find the starting location of each command.

arkypita, 2017-11-09:
> I think you have to analyze the XYZ coordinate and Gcode to figure out which command was active

@winder it could not be done but as you have pointed out is not an easy and reliable method. also you could have wrong coordinates info from filed (remember that they only update every n° msec so you could have an old information)

@chamnit as you say line number report is the most reliable, but it is not practicable for very important reasons. 

I also think about the usage of the buffer information is status report (ie. <Run|....|Bf: 4,83>) to compute how many movements (of which I have already received the ok) are in planned buffer.
Anyway I also doubt that it is reliable for this usage because it is communicated only every n° msec so it could be an old information.

> I'm sure there are other ways

I think the only way to be sure to don't miss a point it is to restart "some line back" from the last OK received. Some line back means 17 movement lines that is the planned buffer size according to your documentation. Of course you risk to repeat some already executed task but you don't miss anything.

> What you are asking for is something called 'program restart' in the CNC world. You have to be very careful about this, especially when dealing with a laser that can cause fires. It's very easy to incorrectly set the g-code modal state and the initial position at the resume point.

You are right! The first test I made was simply to skip all the line before the resume point, and then start pushing remaining lines. You could easily imagine the result :-)

So I have implemented a reading and parsing of the skipped lines (I have all the function already) to rebuild the state of S, F, M3/M4/M5, G90/G91 and position till the restart point. So when the user ask to start from a position I rebuild those states, i move to the restart point with all equipment off, and i restore the situation.

Of course as you already mention I should not forget nothing... for example I have just forgot coolant :-)

@winder 

> Even today you can trick GRBL into reporting the state at an arbitrary line of code by running the program in check mode and making a $G request after sending the lines to be skipped.

This is a nice Idea :-)
If you are right i think I can rebuild the exact state performing $C, run al the previous lines, query $G and parse the result - exit from check mode and use that values for restore the state, instead of gather that from the analysis of the gcode by my routines. (nice idea but a little bit slow)

chamnit, 2017-11-09:
Grbl's check mode $C was original installed with the purpose of adding 'program restart', but I determined that this was better handled by a GUI. However, you don't need to run check mode explicitly. If you know where you stopped, you can request the gcode state $G and $# parameter state before you reset. 

Prior to the restart point, you also need to ensure changes within the $# parameters and any input from a probing cycle for something like a tool change is properly handled.

arkypita, 2017-11-10:
About computing the exact point where to restart from, this is what I finally think:

- If the causing issue is a "cable disconnect" grbl continue executing all command in its buffers, so restart from last command sent.
- If the issue is hard reset, soft reset or grbl hang, grbl does not execute the data in buffer, so restart from 17 positions before the last OK received (the size of planned buffer)

> If you know where you stopped, you can request the gcode state $G and $# parameter state before you reset.

Sometime it is not possible. If the stop is due to a reset (hard or soft) $G is not reliable: better rebuilding state infos from file.

> Prior to the restart point, you also need to ensure changes within the $# parameters and any input from a probing cycle for something like a tool change is properly handled.

Because of my ignorance in cnc world I don't know what does this all means. But if you say that, i believe is also important. Could you explain it better?

- Does not $# be saved in eeprom automatically?
- What is a probing cycle and why I must rebuild its input during a 'program restart'?
- If the program stop i guess that the last tool used is ok for continue the job... why I should handle a tool change?

arkypita, 2017-11-10:
One more question, if I can. I have some problem restoring the modal state of G2/G3
To restore the state I follow this table found in wiki

| Modal Group |  Member Words |
|:----:|:----:|
| Motion Mode | **G0**, G1, G2, G3, G38.2, G38.3, G38.4, G38.5, G80 |
|Coordinate System Select	| **G54**, G55, G56, G57, G58, G59|
|Plane Select	| **G17**, G18, G19|
|Distance Mode	| **G90**, G91|
|Arc IJK Distance Mode | **G91.1** |
|Feed Rate Mode	| G93, **G94**|
|Units Mode	| G20, **G21**|
|Cutter Radius Compensation | **G40** |
|Tool Length Offset |G43.1, **G49**|
|Program Mode | **M0**, M1, M2, M30|
|Spindle State |M3, M4, **M5**|
|Coolant State	| M7, M8, **M9** |

I analyze the file looking for the last command of each group, till the restart point.
Than I send a command with the found value for each group, to restore the modal state of each group.

I see that G0/G1 can be sent also without XYZ parameters with the result in changing the modal state: this is good!

I also see that G2/G3 can not be sent without XYIJ. When I try to do that, i get back an `Invalid gcode ID:26` and this is evil for me :-(

For this reason I can not properly restore the status and resume a job when G2/G3 is used as modal to perform a sequence of arc when gcode is optimized like:

```
G1
X20. Y30.
X10. Y30.
G2 X9.8189 Y55.8487 I520.0261 J16.5689
X10.8904 Y82.787 I614.9038 J-10.969
X12.7298 Y104.07 I434.597 J-26.8386
X15.7268 Y122.7344 I193.8699 J-21.5579
X21.0961 Y140.5441 I107.0443 J-22.5573 <--- i.e. I want to restart from here
X26.8004 Y148.5746 I19.2816 J-7.6552
X34.7303 Y147.1757 I3.3962 J-3.9235
X43.2745 Y129.1779 I-134.2882 J-74.7786
G3 X52.135 Y110.4832 I149.7887 J59.5462
X61.0307 Y108.1798 I5.5429 J3.0773
```

- I rebuild the info "we have G2".
- I also rebuld the info "the previous position is X15.7268 Y122.7344"

If i send the restart sequence

```
G0 X15.7268 Y122.7344 <--- fast move to the previous position
G2 <--- set modal G2
```
I have an error... so the next commands are not executed correctly because the state does not change to G2

[image]

winder, 2017-11-10:
You're right, I noticed this recently as well. In addition to that, if the feed rate hasn't been set yet there is an error for any of the motion commands: `(error:22) Feed rate has not yet been set or is undefined.`

arkypita, 2017-11-11:
Yes, but with the feed rate this is the correct behaviour. Is with G2/G3 that I don't understand if I miss something or it is a grbl issue

arkypita, 2017-11-11:
Here is the list of the commands that give errors if sent individually

[image]

109JB, 2017-11-11:
The errors listed are correct per standards. What you are proposing would be a break from standard and I would disagree with that.

arkypita, 2017-11-11:
@109JB if they are correct per standards I agree with you and with the implementation.
Do you know if there is a particular reason for this behaviour?

I think that I can easily handle this scenario re-adding the missing (optimized) G3/G4 in the very next XYZIJ line where I am restarting from.

note. I am not proposing, I am just asking. (maybe there may be some misunderstanding with my english).

109JB, 2017-11-12:
G01 by definition is a linear interpolated feed rate based move.  When a machine is started up the default feed rate is zero.  You must first define a feed rate before you can implement a G1, whether movement is defined or not..

For G2/G3, you need a feed rate but you also need a center point and an end point. If the end point is ommitted, it is assumed that the end point is the same as the start point, a complete circle.  If the I,J,K words that define the arc center are missing then the G2/G3 can't function.  G2/G3 are kind of different in that while G2/G3 is modal, the I,J,K words aren't, so every time you have a G2/G3 you have to have a I,J,K.   G2/G3 is also different from G0/G1 in that while G0/G1 can be invoked by themselves with no movement implied, a G2/G3  on a line implies an arc move.  Once an arc move is made, the G2/G3 is modal.  As an example

G2 << This line would be an error

G2 I2 << This line is OK in some interpreters but not others. Some interpreters need an X or Y too. Others assume a complete circle and use the start point as the end point.

G2 X2 I1 << This line is OK, provided the end point doesn't pose a radius conflict

G2 X2 I1 << assume this line is followed by These:
X0 I-1  << This line OK and is interpreted just like G2 X0 I-1 since the G2 part is modal.
X2 << This line is not OK because the center point is not defined

arkypita, 2017-11-12:
I think I have understand.
With my words: G1/G2/G3 means "execute a line, cw arc, ccw arc" and only as secondary effect "set the state". So they could be only executed if they are formally valid.

Since an arc is not defined without IJK, any G2 without those params give back the error.

arkypita, 2017-11-20:
Thanks all. Driven by your suggestion i have developed my own "Job Resume after problem" procedure.

closed 2017-11-20 by arkypita

winder, 2017-11-27:
@arkypita @chamnit do you think it's worth making a wiki page that describes this algorithm? Or maybe add it to the end of the Interface page similar to how the Joystick algorithm is documented?

arkypita, 2017-11-27:
@winder 
How to resume from a badly interrupted job is a task that interest only the PC software (streamer) not grbl itself. The way to implement this feature could be very different based on how the program is designed.

However here is a brief description of how "job resume" is implemented in LaserGRBL:

**During the normal file streaming operation:**
a1) keep trace of the last command sent - _line number_
a2) keep trace of the last command managed/answered by grbl (ok or error response) - _line number_
a3) try to detect why the streaming process was interrupted (soft reset by user, disconnect, hardware reset)

The 2nd an 3th step is not really necessary but I have implemented them to suggest the best line number from where restarting from. On disconnection it is possible that all the lines already sent have been executed; on hardware reset the execution may have stopped some lines before the last answered etc..

**When the user press the "send file" button**
b1) recognize that a previous job was not completed
b2) detect if the current position is good or compromised (If there was a hardware reset the reported position is 0,0,0 and  but the laser head could be in any place).
b3) open a dialog form asking for a line number from where restarting from (suggesting a good line number)
b4) suggest to redo homing if the position is compromised 
b4) let the user decide if rerun the job from a specific line, redo homing or not, or start again from the beginning

**When the user decide to resume the job from a specific line**
c1) Parse the whole file until that line and compute:
- the position (xy) from where to restart
- the value of all modal state at the line
c2) Enqueue a sequence of command to:
- redo homing if required by user
- move the laser at the position identified in previous step
- restore the modal state computed in previous step
c3) resume the streaming from the next line and push all the subsequent commands