gnea/grbl #306: Porting to ATmega2561 Intrrupts

WD24, 2017-11-12:
Has anyone ported GRBL on to an ATmega2561?

As far as I can see the ATmega2561 has only 1 port which handles pin change interrupts (PCINT) Unlike the ATmega2560 and the ATmega328.  GRBL needs two ports one to handle the hard limits and one to handle the control pins. 

The Atmega2561 does have other interrupts (INT). From what I understand these interrupts only triger when they are pulled low. whilst the pin change interrupts trigger if the pin changes from low to high or it changes fom high to low.

Can I use these INT instead of the PCINT ? if so how would you impliment this?

Schildkroet, 2017-11-12:
Read the datasheet section 15 external interrupts:

> The External Interrupts are triggered by the INT7:0 pin or any of the PCINT23:0 pins. Observe that, if enabled, theinterrupts will trigger even if the INT7:0 or PCINT23:0 pins are configured as outputs. This feature provides a wayof generating a software interrupt.The Pin change interrupt PCI2 will trigger if any enabled PCINT23:16 pin toggles, Pin change interrupt PCI1 if anyenabled PCINT15:8 toggles and Pin change interrupts PCI0 will trigger if any enabled PCINT7:0 pin toggles.PCMSK2, PCMSK1 and PCMSK0 Registers control which pins contribute to the pin change interrupts. Pin changeinterrupts on PCINT23:0 are detected asynchronously. This implies that these interrupts can be used for wakingthe part also from sleep modes other than Idle mode.The External Interrupts can be triggered by a falling or rising edge or a low level. This is set up as indicated in thespecification for the External Interrupt Control Registers – EICRA (INT3:0) and EICRB (INT7:4). When the externalinterrupt is enabled and is configured as level triggered, the interrupt will trigger as long as the pin is held low

WD24, 2017-11-12:
When you compile GRBL in the Arduino IDE, for the ATmega2561 it creates errors saying it does not recognise PCMSK2 and PCINT2.

If you look at the graphic in the data sheet (section 1 Pin Configurations) for the pin layout of the 64 pin ATmega1281/2561(Figure 1-3.) it only has PCINT0-7 on pins PB0 - 7 and PCINT8 on PE0 (which is also RX0)

As apposed to Figure 1-1 for the 100 pin ATmega640/1280/2560 which has PCINT0-7 on pins PB0 - 7, 
PCINT 8 on PE0, PCINT 9-15 on PJ0-6 and PCINT 16- 23 on PK0-7

SO on the ATmega2561 there are not enough ports avalible for all the control pins and the limit pins. 

Am I missing some thing? 

Can you use INT pins instead of PCINT and if so how?

chamnit, 2017-11-12:
This issue is closed, because it is unrelated to the supported 2560 processor.

closed 2017-11-12 by chamnit