gnea/grbl #308: Spindle speed measurement

goloplo, 2017-11-15:
Can pin A4 on Arduino be used with an it sensor to count RPM of the spindle?

Thank you.

chamnit, 2017-11-15:
It sure could, but it won't be on the 328p version of Grbl due to flash, RAM, and even performance constraints. The interrupt has to be serviced every sensor tick and would quickly eat up cycles at high rpms.

closed 2017-11-15 by chamnit

goloplo, 2017-11-15:
Can STM32 handle it? If yes is just using the probe script as reference?

graphtjb, 2019-07-10:
I agree with the though about spindle speed eating up clock cycles, but frequency divider could slow up the that speed. 
https://www.electronics-tutorials.ws/counter/count_1.html
So the reading would be 2^n. We could account for this 2^n factor in programming. 10 d flip flop circuits could slow up the rpm reading from a encoder about a 1000 times. Make a Mhz signal into a khz signal.