gnea/grbl #309: Homing X-Y Axis / limit switch issue.

MaartenHeuv, 2017-11-26:
Basic setup:
- GRBL 1.1f
- XYZ axis 
- 1 limit switch per axis
- NO switch
- $5=0 (limit pins invert, bool)
 
Normal operation is running fine: directions okay, mm/step okay etc.
Z_axis behaves as mentioned in https://github.com/gnea/grbl/wiki/Set-up-the-Homing-Cyclend then moves 1mm? into the active direction.

After the Z_axis I get; ALARM 8
Therefore I compiling version with "HOMING_SINGLE_AXIS_COMMANDS" enabled

Both X_axes and Y_axes behave same:
Case middle normal travel range: Moves 2 steps into switch direction --> message ok
Case within 2 steps from limit switch: Moves 1 step into switch direction, limit switch, ---> ALARM 8
Case limit switch ia allready reached: ---> ALARM 8

If I the limit switch to NC (By soldering):
allways Moves 1 step into switch direction, limit switch, ---> ALARM 8  

What is wrong here ?

Beside this issue, I want to mention: this GRBL project is a great job!!!!_

109JB, 2017-11-27:
A few questions and comments.

Many limit/homing switch problems are the result of electrical noise causing false triggers. With that in mind:
1. Do your XY limit switch wires run alongside your stepper wires or any other wires that could be carrying any current?
2. Do the limit pin circuits have any filtering? IE: capacitors, resistor/capacitor combo (Low pass filter), opto-isolators, etc.
3. Are either the stepper wires or the limit switch wires shielded cable and are they properly grounded at only one end in the electronics cabinet/box?
4. Do you have an electronics cabinet/Box? A metal box is best and provides shielding for the stuff in the box.
5. Is the stepper power supply properly grounded and how close is it to the Arduino?

Also, out of curiosity, you mention that when the error happens it moves only 2 steps or only 1 step.  How are you determining that this is the case?  During homing the status reports are suppressed

MaartenHeuv, 2017-11-27:
Dear 109JB,
Thanks for your quick reaction.

Bingo, solved.

1. I used shielded wires. with seperate ground for switches.
2. I "decoupled" the 3 inputs with 10nF close to the AVR, HOWEVER after a recheck I found that the expected AVR grounding point was NC.......
After connecting this to the real AVR ground it was soved.

"2 steps"
Perhaps my explanation was ambiguous: I observed it visual, 2 "step" of roughly 1mm, so in stepper language 2 burst of  133 steps, both in the home direction.

Thanks again.

closed 2017-11-27 by MaartenHeuv