gnea/grbl #310: Homing in three cycles

3kmch, 2017-11-26:
By now I used our GRBL controlled Stepcraft router without homing. So if something went wrong I completeley lost the workpiece position. Last week I decided to change that. ;-)

Unfortuneately the Stepcraft fires no specific signals, if the reference switches are hit. That means GRBL just gets the same signal for switch Z, X and Y. No problem so far. Instead of a two cycle homing (first Z, then X and Y as default) I have to change it to a three cycle homing (each axis separetely).

Default settings in config.h:
(see also https://github.com/gnea/grbl/blob/master/grbl/config.h#L105)
```#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))
// #define HOMING_CYCLE_2
```

I changed these lines to:
```
#define HOMING_CYCLE_0 (1<<Z_AXIS)
#define HOMING_CYCLE_1 (1<<X_AXIS)
#define HOMING_CYCLE_2 (1<<Y_AXIS)
```

That was easy! But nothing changed! The router keeps homing in two cycles. :(

What I tried so far:
- deleted all the precompiled files unter "C:\compiled\libraries\grbl" (working on W10)...
- deleted all the GRBL files and started from scratch...

For me it looks like my bloody computer uploads the unaltered code. How can I verify wether that's a fact or not?

109JB, 2017-11-27:
Are you saying that your machine only uses one pin for all 3 axis limits?  If so, then you have to also change the pin assignments for the limits so that grbl is looking at the correct pin when homing. I did this on my last machine build and the changes I made are in the link below.
https://github.com/gnea/grbl/issues/37

3kmch, 2017-11-27:
Yes, that's right. But at the moment all three input pins are connected. So the signal is led to all pins. In this case I don't thing I have to change the pin assignment.

What I don't understand is that the machine moves still in both axes (simultaneously in X and Y).

3kmch, 2017-11-27:
Problem solved! I'm very happy but it is also a bit embarrasing:

As an Arduino newbie I copied the code to the desktop and did all the changes there instead of the place, where all the arduino libraries are located. :(

Sorry

closed 2017-11-27 by 3kmch