gnea/grbl #311: disable z option in laser mode

sinfocomp, 2017-11-27:
hello, would it be or is it possible to disable z while in laser mode, I am using v 1.1 in my laser machine and cannot use limit switch on x, y because homing fails because there is no z hardware at all also by disabling z software could either ignore all z movements or use the z height value as pwm laser power value to engrave or cut according to the z value
thanks
Ed.

votrungchi108, 2017-11-27:
I guess that is what you looking for. 
https://github.com/gnea/grbl/wiki/Frequently-Asked-Questions#how-do-i-configure-my-homing-cycle-to-just-do-2-axes-or-i-have-a-pen-plotter-and-dont-have-a-z-axis

sinfocomp, 2017-11-27:
would partially work, but since  a $ setting is already implemented for laser mode it would be nice if this disables or enables z axis, my machine has dual gantry setup on one I have a laser head and plasma cutter, the other one has a router as spindle, so recompiling or reflashing each time I want to switch its kind of a pain
thanks

biasedlogic, 2017-11-27:
Why would you assume every laser is two axis only?

sinfocomp, 2017-11-27:
I'm not assuming that, this is why I ask if  a $ config option is possible, I have been thinking on installing a z axis on it for focusing the laser and the possibility of controlling engraving line width

chamnit, 2017-12-07:
You'd be better off getting a controller for each machine you have. It sounds like you are moving it around to machine to machine. That is much more of a pain than having to reflash. Adding such an option is not possible due to flash and memory restrictions. The 328p is literally completely out of room for certain critical configurations. 

As for your question on using the z-axis for laser power, it's been done before and works ok. It tends to become problematic when you have a laser and a z-axis. However, I still don't see what the problem is, because Grbl v1.1 already addresses any performance issues with using the S spindle speed to control laser power.

closed 2017-12-07 by chamnit

bcabral, 2021-08-28:
Hello. When I use hard limits NC in grbl 1.1h, it complains about not having a Z (triggered always) in a openbuilds ACRO with a laser.

Should I short the pins or is there any software config (perhaps limits mask?) to disable Z checking? I already did the HOMING mod in FAQ quoted above

Thanks

langwadt, 2021-08-28:
@bcabral change the homing cycle (line 105/106 in config.h) and rebuild the code

bcabral, 2021-08-28:
Hello, thanks for answering

The first thing I did was to comment lines 105 and 106 in config.h and enable line 110, then reflash.

    105 //#define HOMING_CYCLE_0 (1<<Z_AXIS)                // REQUIRED: First move Z to clear workspace.
    106 //#define HOMING_CYCLE_1 ((1<<X_AXIS)|(1<<Y_AXIS))  // OPTIONAL: Then move X,Y at the same time.
    107 // #define HOMING_CYCLE_2                         // OPTIONAL: Uncomment and add axes mask to enable
    108
    109 // NOTE: The following are two examples to setup homing for 2-axis machines.
    110 #define HOMING_CYCLE_0 ((1<<X_AXIS)|(1<<Y_AXIS))  // NOT COMPATIBLE WITH COREXY: Homes both X-Y in one cycle.

Here is what I did next:

$5=1 because my switches are NC
$20=0
$21=0
$22=1 to enable homing
$23=3 to home toward me instead of the other way around

Then I turn verbose on in UGS. I can see Z "triggered" always and X and Y only shows when I press the endstops.

[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:Z>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:XZ>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:XZ|WCO:-499.000,-499.000,0.000>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:Z|Ov:100,100,100>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:Z>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:Z>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:YZ|WCO:-499.000,-499.000,0.000>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:YZ|Ov:100,100,100>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:YZ>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:Z>
[verbose]<Idle|MPos:-332.440,0.000,0.000|FS:0,0|Pn:Z>

If I try to home or try to move, nothing happens

ok
>>> $H
>>> G21 G91 G0  Y100
>>> G91 G0  X100
ALARM:9
ok

If I short Z, home sorta runs. Sometimes X and Y, sometimes X starts then stop without reaching the endstop, while Y continues until reach the endstop, and sometimes only Y moves

If I turn $21=1 I got all kinds of errors

(soft reset)
>>> $X
[MSG:Caution: Unlocked]
ok
[verbose]<Idle|MPos:-499.000,-499.000,0.000|FS:0,0>
ALARM:1
[MSG:Reset to continue]

(the ALARM is so fast that I cannot press $H after $X before it shows)

I am a bit lost here.

The homing_cycle define shouldn't turn off Z ?

Is the Z short the correct solution ?

Why $21=1 is behaving like this ?

I have two Y and two X endstops (NC) in series, uno and cnc shield v3, grbl 1.1h. Without endstops, the machine runs perfectly

>>> $
[HLP:$$ $# $G $I $N $x=val $Nx=line $J=line $SLP $C $X $H ~ ! ? ctrl-x]

Cheers
!3runo