gnea/grbl #312: Request for separate 'ok' confirmation for gcode receipt

Neex101, 2017-11-27:
Hi,
Love grbl 1.1 by the way.
I'm writing a custom serial interface and having to do some intricate work in the code to separate out the "ok"s from gcode sends and "ok"s from other realtime commands (e.g. status). Since maintaining the g-code buffer relies on checking these "ok"s, I would have found it heaps simpler (and more reliable for my code's perspective) if gcode receipt confirmation was different to that for the other commands. e.g. 'gk' etc.
Not a showstopper obviously, I'm sure I'll get around it.
Out of interest, is this an easy thing to change the message in the grbl library for a custom build? I haven't messed with that before.
Many thanks for the superb work.
Best,
Ed

chamnit, 2017-11-27:
@Neex101 : Thanks for the input. I plan on updating the comm protocol in the next major version and will include something like what you propose. But as it is now, it can't change. There are too many dependencies.

closed 2017-11-27 by chamnit

winder, 2017-11-27:
@Neex101 you might be mistaken, grbl does not respond with "ok" for real time commands (`?`, `0x90`, `0x91`, etc).

You might be thinking of `$G` and `$$`, but those are not real-time commands so it would be a little unusual to mix them into a gcode stream. You also need to be extra careful not to send anything after those commands because they may read/write EEPROM which temporarily prevents grbl from reading the serial bus, causing lost packets.

Neex101, 2017-11-28:
@winder hi, thanks for the caution on the setter commands. However, I'm definitely get an "ok" after the status report, e.g.:
> ?
>>> <Idle|MPos:0.000,0.000,0.000|FS:0,0>
>>> ok
I am running grbl v1.1. If you're not getting the 'ok', does that mean you have been able to disable it?  I may have missed how to do this...

langwadt, 2017-11-28:
@Neex101 are you sending a newline after the "?", you shouldn't

Neex101, 2017-11-28:
Thanks guys, that was my problem. The final serial write function was sneaking in a '\n'. My bad - this was very clear in the docs too. Great, parsing will be a breeze!