gnea/grbl #313: Compensate not 90 degree axis

Ravenbs, 2017-11-28:
Feature request:
All major vendors of CNC machines offer a software compensation for geometric machine errors.

Most important is to calibrate all three axis direction.
So for example the direction of x axis could be 0.99x and 0,01y

Would this possible?

lalo-uy, 2017-11-28:
It will be easyer to impement on the GUI

Ravenbs, 2017-11-29:
Do you know a GUI where it is implemented?

I think it is not complicated in GRBL.
From my understanding it is one matrix multiplication on the output set of coordinates.
All misaligned axis, shifted axis,... can be molten down to a single 4x4 matrix (like in 3d Gaming, the P Matrix).

swarfer, 2017-11-29:
a 4x4 matrix multiply is a LOT of clock cycles on an 8bit microprocessor that does not have a hardware floating point processor.   Doing this in realtime for every co-ordinate within GRBL is to going to work on a processor (the atmega328p) that is already at capacity.

But, for a PC with a floating point processor to do this BEFORE sending the co-ordinates to GRBL is quite possible.
So the right place to do it is inside a GUI.  Pick one of the open source GUI's and implement this and submit a pull request (-:

X3msnake, 2017-11-29:
I think BCNC and chillipepr already have a xy(z) plane compensation
feature. This can probably be used with little modification to do skew
compensation if it is not implemented already.

I have read in the BCNC git something about implememting backlash
compensation, that might intetest you too.

X3msnake, 2017-11-29:
It is already there this was 2yrs ago dunnonif it is implemented already
today

"If we assume that the machine is not properly calibrated and not squared
we can find the transformation matrix between your pcb reference system and
the machine with minimum 3 points, the more the better would become. This
is the aim of the "Square" tab inside the "Probe" page. For the moment is
not implemented, but in principle is easy to do."

https://github.com/vlachoudis/bCNC/issues/140

Ravenbs, 2017-11-29:
Thank you for this hint - but it is a different Feature.

BCNC can compensate z offsets by multiple probes very well.

The Square tab is if you place an object again on the table, you can rotate the g-code to fit the material on the table.
But the Axis are still asumed 90° only the complete Gcode is rotated.
For example for drilling a PCB after you milled it and had to remount it on the table.

chamnit, 2017-11-29:
@Ravenbs : Not going to happen on the 328p AVR version of Grbl. No room, CPU power, or memory. This feature is already on my list of things to do, but it isn't a priority. Mainly due to the fact that defining corrections can get quite complex, especially if it's going to be generalized for everyone, and there are other pressing matters.

closed 2017-11-29 by chamnit