gnea/grbl #314: Signifigant speed differnece between M3 and M4 in laser mode

NuttCorp, 2017-11-29:
I am setting up a CNC shield with an Arduino Uno and I have everything working with the exception of M3 being significantly slower. I am using the constant power mode for this project and it works for the most part but it is VERY slow. I understand that because it is in constant power mode it doesn't compensate laser power dependent on the speed but I have my acceleration turned way up. I thought that the 2 power modes only effected the PWM signal to the laser but if I generate 2 gcode files one using M3 and the other using M4, it take 3x longer for M3 to run. Same everything except the spindle command.
M3 = Elapsed time: 0:09:06
M4 = Elapsed time: 0:02:49

[gcodeM4.gcode.txt](https://github.com/gnea/grbl/files/1512601/gcodeM4.gcode.txt)
[gcodeM3.gcode.txt](https://github.com/gnea/grbl/files/1512602/gcodeM3.gcode.txt)

chamnit, 2017-11-29:
Your buffer is starving because of how small the motions are. They are 10-20 **microns** in length. About an order of magnitude smaller than what is recommended. If you want that small of detail, you'll have to significantly reduce the feedrate.

FYI, Grbl performs very well on the slow AVR chip when compared to firmwares on faster chips. Grbl on fast LPC chips blazes. So, don't expect to get better results on other platforms, other than professional lasers, until Grbl's next version is released on ARM chips. There is a focus on improving laser performance.

closed 2017-11-29 by chamnit

NuttCorp, 2017-11-29:
Ok. I was just curious because I get the performance I want with the M4
command but dont want dynamic power, and with M3 for the constant power it
triples the operation time. I just dont understand why changing the pwm
profile for the laser/spindle affects the performance of movements. I am a
newbie so please dont go crazy on me. I am just try to understand.

chamnit, 2017-11-29:
I'm not sure what you mean "by go crazy on me". I simply told you the problem, the solution, and how to make it better. If being direct and honest is crazy, lord help us.

NuttCorp, 2017-11-29:
I wasnt referring to what you had said but to your possible response to my
reply of not understanding.