gnea/grbl #315: CoreXY not work in grbl 1.1f

marshalab, 2017-12-05:
Hello/Privet/Nihao ! :)
I am create https://www.thingiverse.com/thing:2401997 2D printer and want setup 2w5 laser on ware. Mechanix is good, but with electronic I've been trying all day, not work... I am using Arduino nano 328p with diterct download firmware over USBASP, because after enabling the option CoreXY firmware so BIG for nano. 
If download grbl 1.1f by default, my hardware is connect, motor is rotate, but wrong. Than try #define COREXY option - after downloading - connection is not established ! The firmware does not respond on any bitrate... 
Where it does not start.

marshalab, 2017-12-05:
ааа..... in the limits.c
if comment strings...
              #ifdef COREXY
                if (idx==Z_AXIS) 
			{ axislock &= ~(step_pin[Z_AXIS]); }
                else 
			{ axislock &= ~(step_pin[A_MOTOR]|step_pin[B_MOTOR]); }
              #else
                axislock &= ~(step_pin[idx]);
              #endif

connect is good.
mining deeper...

marshalab, 2017-12-05:
Uraaaa ! I have connect!
in limits.c
        for (idx=0; idx<N_AXIS; idx++) {
change on
        for (idx=0; idx==N_AXIS; idx++) {

and i have connect. Testing further.

chamnit, 2017-12-05:
@marshalab : CoreXY works. You don't need to alter the source code. Make sure you editing the config.h file in the Arduino library folder. Not your project or download folders.

closed 2017-12-05 by chamnit

marshalab, 2017-12-05:
? How its may work without conntction ? Try connect to Arduino with enable COREXY. 
Dont delete this masages, maybe people need it. Pls/

chamnit, 2017-12-05:
Nothing gets deleted. An issues thread can be closed and still be responded to and be searched for. Don't worry.

If you have connection failures, it's much more likely that its something to do with your Arduino or setup, not the Grbl firmware. I suggest you disconnect everything from the Arduino and start doing some simple tests to isolate where the problem is coming from.