gnea/grbl #316: Why limit switch automatic Worked without pressing any limit switch?

lantechbd, 2017-12-07:
Why limit switch automatic Work without pressing any limit switch, Please help me to solve this problem.

dungjk, 2017-12-07:
You may got noise issue.
1. Keep limit switch cable away noise source (stepper motor wires, fan wire, power wire, laser pwm or other pwm source).
2. Check if limit circuit has noise filter (common PI filter). If not you may build one for each limit switch, it's very simple.
[image]

lantechbd, 2017-12-07:
dungjk// Thanks

closed 2017-12-07 by chamnit

lantechbd, 2017-12-07:
TP1 and Detect whats that...
Detect means = Optocupler???

chamnit, 2017-12-07:
Please read the wiki page on limit switches.

lantechbd, 2017-12-07:
Ok, Thank you.

wlaw, 2017-12-08:
Md Kawser Ahmed,

There is soooo much information on this false tripping of the limit
switches.

It usually comes down electrical static noise suppression.

here is a link to start you in the correct direction.

https://github.com/gnea/grbl/issues/96

Kind Regards

lantechbd, 2017-12-08:
Great! Thanks from my heart for your help.

lantechbd, 2017-12-08:
[image]

is this true circuit...