gnea/grbl #317: 2 y steppers

ve3sqb, 2017-12-09:
Is there an easy way to make the z output into a Y2?  I'm thinking in the Grbl setup or the gcode. Also verify that pin 11 is the 5 volt ttl output.   Thanks Al

biasedlogic, 2017-12-10:
1 Split wire?
2 multimeter?

ve3sqb, 2017-12-10:
I don't have the board yet so I can't check the pins. Split wires would push the stepper driver over the 2 amp limit and steppers don't come in matched pairs so I've seen a stronger one capture all the voltage and the weaker just chugs along. The unused z axis gives me another driver so I can run two 1.5 amp steppers.

biasedlogic, 2017-12-10:
Split dir/step/enable, not the power wires...

ve3sqb, 2017-12-10:
and how do I split?

biasedlogic, 2017-12-10:
Well, I'm freshly all out of crystal balls and my tarot deck is acting up, so can't help really about your hardware set, which you conveniently kept secret. In my box you just jumper the A axis control signals to whatever you like...

jahnj0584, 2017-12-10:
You run the one dir/step/enable pin for Y to both motors. Split the cable
however you see fit. I used a terminal block. Power supply + driver are
seperate per motor, ideally.

2* 1.5A steppers do not draw 3A at the same time. They are rarely ever
under full load

ve3sqb, 2017-12-10:
If you read the question I was looking for a software answer, not hardware. Have a nice day tomorrow because you seem to be off today

closed 2017-12-10 by ve3sqb

chamnit, 2017-12-10:
@ve3sqb : There is no easy way to alter Grbl's stepper output to do dual y-axis in software, like you asked. It's an interesting idea though. What exactly do you need this for? Do you need independent behavior between the two y-axis motors? Otherwise, if you don't care, you can simple share the y-axis pin between both motors. This is a very common solution.