gnea/grbl #318: how to control servo motor speed

bijlevel, 2017-12-10:
I'm using the software with a 4xiDraw plotter (drawing machine). It works perfectly, except the high speed of the servo motor causes the plotter pen to bump quit hard on the surface, which causes unwanted line artifacts. I know I can control the amount of rotation with a M3 Sxx command, but that only controls the distance of the pen from the drawing surface, not the downward or upward speed. How do I accomplish this, if even possible? Thanks in advance!

Kind regards,
Sjoerd

chamnit, 2017-12-10:
@bijlevel : Unfortunately, there isn't a way to control the acceleration of a servo at this time. I'll make note of this need for future development though.

X3msnake, 2017-12-10:
@bijlevel <https://github.com/bijlevel>

Maybe you can do this via hardware

What Hardware are you using to drive the servos?

bijlevel, 2017-12-11:
@chamnit: thanks anyway, maybe there is a solution via the hardware.
@X3msnake: I'm using a CNC Shield For Arduino (GRBL Compatible), mounted on an Arduino Uno. The servo motor (just one) is connected to the Z+ of the end stops (X+, X-, Y+, Y-, Z+, Z-). The servo type is SG90 RC Micro Servo. Follow the link for a picture of the shield:[http://store.thingibox.com/en/electrical/196-cnc_shield.html](url)

bijlevel, 2017-12-11:
@chamnit: thanks anyway, maybe there is a solution via the hardware.
@X3msnake: I'm using a CNC Shield For Arduino (GRBL Compatible), mounted on an Arduino Uno. The servo motor (just one) is connected to the Z+ of the end stops (X+, X-, Y+, Y-, Z+, Z-). The servo type is SG90 RC Micro Servo.

chamnit, 2017-12-11:
You could program in g-code a set of M3 Sxxx commands with a G4 Pxxx dwell in between. Just slowly ramp the servo signal. This could be accomplished with a simple find and replace function of a good text editor. 

For something non-blocking (synced with motion), you'd need to be comfortable with programming and not using the spindle commands for anything but your servo. If it were me, I'd alter the spindle_control function to generate a time-varying output and use a timer (watchdof or timer2) with an interrupt to manage how the output is ramping.

biasedlogic, 2017-12-11:
You can use another Arduino to sort of emulate a stepper controller. You could use a servo library to output the servo signal, keep the last position in a variable, and then watch two pins (pick yourself which you use) in main loop: if DIR pin is high, ticking STEP pin should increase position variable (up to some max value), if DIR pin is low you decrease position value down to zero. At zero you set some other pin to low, so that you can wire it to your grbl controller as end stop signal.

Now you have a full blown synchronized motion control on a servo

bijlevel, 2017-12-11:
@chamnit The g-code programming sounds feasible to me (and a creative solution, for sure!). I will try that first. Thanks!
I assume the second solution requires altering the spindle_control.c file. For now that is a bit (or way ;-)) above my head, sorry.

bijlevel, 2017-12-11:
@ biasedlogic thanks for the reply and help! As mentioned in my previous reply to @chamnit I'm not (yet) very skilled in this kind (or for that matter other kinds ;-)) of programming, so I will try the g-code solution first.

X3msnake, 2017-12-11:
Cool, post back the solution you end up using :)

;)

bijlevel, 2017-12-12:
Well, I found out the problem has to do with the way the plotter is build. The distance of the plotter pen to the plotting surface is not uniform over the plotting area. If it was uniform, then carefully positioning the pen in the penholder in the way it touches the plotting surface just enough to draw, would solve the problem. The arm to which the penholder is attached is not firm enough, so when stretched out all the way, it bends more toward the plotting surface then when retracted. So the solution must be a mechanical one, using at least more robust materials or maybe some kind of spring mechanism for the pen... Thanks anyway for the input and help!

neilferreri, 2017-12-12:
I'd recommend using gravity to lower the pen and the servo to lift it.
Don't push the pen into your surface, just allow it to drop.

X3msnake, 2017-12-12:
https://www.thingiverse.com/thing:1105152

bijlevel, 2017-12-12:
@ neilferreri my mechanism for pen up and down is controlled by the servo motor, or to put it different the pen is attached to the servo. So when I use M3 S0 it pushes the pen down, but because of the non uniform distances from the plotting area, sometimes this push force is too much and causes the pen to bump into the surface and draw unwanted (small) lines because of glitches.

@X3msnake that looks promising! I will look into it or maybe inspired by it, using some kind of spring mechanism. Thanks!

neilferreri, 2017-12-12:
@bijlevel I've designed and built a few simple drawing robots.  Most of the issues I've had are eliminated by having the servo lift the pen, but using a mass to hold the pen down.  Springs work on robust machines, but they will still create too much down force much of the time, especially on smaller, less rigid, machines.  Basically, when the pen gets lowered the servo arm gets out of the way allows it to drop, it's free to follow the contours of your surface, and then the servo lifts it again.

X3msnake, 2017-12-12:
@neilferreri
That's a nice tip too, one can make a wheight sistem to configure
"pressure"

bijlevel, 2017-12-12:
@neilferreri I will certainly look into this solution! It requires a redesign of the penholder mechanism, but that seems not so difficult. I have already seen some examples. Thanks for your reply and help!

closed 2017-12-13 by bijlevel