gnea/grbl #319: UGS is showing this [Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0> all the time.

Larepera, 2017-12-10:
Hi all!

I found this amazing firmware while looking for an affordable CNC controller and GCode interpreter. Thank you very much to the authors and contributor for bringing this marvelous piece of code available for all!

I used Galil, Fagor and Siemens in other applications but I was interested in a really affordable option intended for a more wide public.
I was very exited by the reading and went to shopping next day to buy an Arduino UNO board to start playing with it.

All the start up is very straight forward so no comments on it. Case is that I had just the tiny Arduino board on the desk connected to a PC by a EMI protected USB cable and the UGS running and started to "move" the axis and sending some commands. All worked fine, but I discovered an unchecked box that will let me jog the axis with the keypad, so I enabled the option and started to push the keys to see what happen... and all went great until I kept the Y+ key pressed for a while. Suddenly the Alarm changed to red and the TX and RX LEDs on the board start to blinking and the verbose shows a MPos at 0,0,0 in a continue scroll. I was not able to stop this anymore. I killed the Allarm... nothing happened. I reloaded the firmware.. no way. I erase the GRBL folder, created a new one, downloaded all the software again, installed them, added the GRBL library in the Arduino IDE and downloaded the GRBL Firmware to the board using XLoader... Nothing changed. Every time I connect the Arduino to the USB I have those LEDs blinking and the massage with the zero position scrolling in the monitor.
Lastly and as an ultimate option, I bought a brand new ATmega 328P-PU, put it on the board, burned the bootloader, and started as a new one and... do you know what? Yes! those LEDs are still blinking and the verbose output scrolling!!
I was not able to find any variable that might retain a value to keep this situation on. I even loaded an EEPROM eraser to clean the processor memory but no luck with this neither. So, if the event is not saved in the ATmega328, nor in the PC disk (I deleted all the GRBL related directories twice) where has been this alarm event stored. What I should make to restore the board (or entire system Ardino + GRBL) to its initial stage and behavior?
It seems to me that it will work OK when I assemble the machine and the controller can find the limit switches and perform the homing cycle but in the meantime I am struggling with all this.

Thank you very much, in advance, for all the help and advice you can provide to me. It will be much appreciated!

PS. Here is my configuration and the output I am talking about:
( [Detallado] is [Verbose] in Spanish)

Grbl 1.1f ['$' for help]
>>> $$
$0 = 10    (Step pulse time, microseconds)
$1 = 25    (Step idle delay, milliseconds)
$2 = 0    (Step pulse invert, mask)
$3 = 0    (Step direction invert, mask)
$4 = 0    (Invert step enable pin, boolean)
$5 = 0    (Invert limit pins, boolean)
$6 = 0    (Invert probe pin, boolean)
$10 = 1    (Status report options, mask)
$11 = 0.010    (Junction deviation, millimeters)
$12 = 0.002    (Arc tolerance, millimeters)
$13 = 0    (Report in inches, boolean)
$20 = 0    (Soft limits enable, boolean)
$21 = 0    (Hard limits enable, boolean)
$22 = 0    (Homing cycle enable, boolean)
$23 = 0    (Homing direction invert, mask)
$24 = 25.000    (Homing locate feed rate, mm/min)
$25 = 500.000    (Homing search seek rate, mm/min)
$26 = 250    (Homing switch debounce delay, milliseconds)
$27 = 1.000    (Homing switch pull-off distance, millimeters)
$30 = 1000    (Maximum spindle speed, RPM)
$31 = 0    (Minimum spindle speed, RPM)
$32 = 0    (Laser-mode enable, boolean)
$100 = 250.000    (X-axis travel resolution, step/mm)
$101 = 250.000    (Y-axis travel resolution, step/mm)
$102 = 250.000    (Z-axis travel resolution, step/mm)
$110 = 500.000    (X-axis maximum rate, mm/min)
$111 = 500.000    (Y-axis maximum rate, mm/min)
$112 = 500.000    (Z-axis maximum rate, mm/min)
$120 = 10.000    (X-axis acceleration, mm/sec^2)
$121 = 10.000    (Y-axis acceleration, mm/sec^2)
$122 = 10.000    (Z-axis acceleration, mm/sec^2)
$130 = 200.000    (X-axis maximum travel, millimeters)
$131 = 200.000    (Y-axis maximum travel, millimeters)
$132 = 200.000    (Z-axis maximum travel, millimeters)
ok
>>> $G
[GC:G0 G54 G17 G21 G90 G94 M5 M9 T0 F0 S0]
ok

[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>
[Detallado]<Idle|MPos:0.000,0.000,0.000|FS:0,0>

winder, 2017-12-10:
You're using the spanish localization so I'm not sure the exact option names. But this "verbose" output is enabled for the UGS console which is why you're seeing these messages.

If you are using the classic GUI un-check a checkbox above the console, if you are using the platform right-click the console and uncheck the option.

jahnj0584, 2017-12-10:
When I had verbose on it also kept scrolling and sending the useless
(verbose) messages and information.

The rx and tx lights flicker with the pulses sent to move an axis. I'm
pretty sure they blink when idling as well. If they are a solid light you
have an issue...

Larepera, 2017-12-10:
Thanks for your comments!
Yes, I know that I can disable the verbose output, but the RX and TX LEDs will be blinking anyway.
They are blinking at idle, yes. I just would like to be sure that it is a normal stage...

Thanks again!

winder, 2017-12-10:
It is normal

Larepera, 2017-12-11:
Well, I am not absolutely sure because the comm channel is transmitting an receiving all the time just to display the position over and over...
Also, although I can not remember so certainly, I believe that, when I started the system first time, the RX and TX LEDs only blinked at the time of sending an instruction, not constantly.
Thanks anyway for your answer!

winder, 2017-12-11:
UGS polls UGS for status constantly. You can change the status interval in the settings if you need the RX and TX LEDs to blink at a different speed...

Larepera, 2017-12-11:
Interesting...! So, winder, would you say that this was the very first stage at the time of the start up of the system?

winder, 2017-12-11:
After you connect to GRBL UGS starts polling it for status. The only time it stops is while you are homing or after you disconnect

Larepera, 2017-12-11:
Oh yes, I agree. I verified that along all the test I did. The disconnection stage is obvious because no transmission will be performed in that stage anymore until you reconnect again.
Just would like to add that the situation is the same using the GRBL Panel software. 
By the way, polling for status all the time is the only way to know (by the software) what is happening with the hardware. I just would like to know from other users if their TX and RX lights are flashing all the time. I almost sure that this was not the initial behavior.
Thanks again for your comments Winder. Much appreciated!

closed 2017-12-11 by chamnit

SherifAbouZeid, 2023-01-04:
I do have almost the same issue when connected and console gives this messages
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0|Ov:100,100,100>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>
[verbose] <Idle|MPos:0.000,0.000,0.000|Bf:15,128|FS:0,0>

Is that an issue with connection or is that the board is not working with my USB connection correctly?
Do need some help .... thank you !!